kismetwireless/kismet #7: capture_hackrf_sweep fails to compile

detlef-bloch, 2017-07-25:
I just tried to compile the latest git master version, but:

gcc -Wall -g -pthread -I/usr/include/libnm -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -I./msgpack-1.4.1/include   -pthread  -fopenmp -I/usr/include/libnl3 -c capture_hackrf_sweep.c -o capture_hackrf_sweep.o 
capture_hackrf_sweep.c:132:24: error: ‘SAMPLES_PER_BLOCK’ undeclared here (not in a function)
 uint32_t num_samples = SAMPLES_PER_BLOCK;
                        ^
capture_hackrf_sweep.c:134:22: error: ‘MAX_SWEEP_RANGES’ undeclared here (not in a function)
 uint16_t frequencies[MAX_SWEEP_RANGES*2];
                      ^
thus,

Makefile:304: recipe for target 'capture_hackrf_sweep.o' failed
make: *** [capture_hackrf_sweep.o] Error 1

With kind regards

Detlef Bloch

kismetwireless, 2017-07-25:
You've got a very old libhackrf; fixed in 72dbb910b8dfa0a0e1a8f7e0741b0eae2cea5462 which checks the contents of the header

closed 2017-07-25 by kismetwireless

detlef-bloch, 2017-07-26:
Hi Mike,
you were right, in a way. The "System" is an old Laptop:
Computer:
  System Info:
    Manufacturer: "FUJITSU SIEMENS"
    Product: "LIFEBOOK S6120"
  Processor Info:
    Manufacturer: "Intel Corporation"
    Version: "Intel(R) Pentium(R) M processor"
    External Clock: 400 MHz
    Max. Speed: 1400 MHz
    Current Speed: 1400 MHz
  BIOS Info:
    Vendor: "Phoenix/FUJITSU"
    Version: "Version 1.22"
    Date: "08/25/2003"
whereas the OS is new:
System:
    Distributor ID:	LinuxMint
    Description:	Linux Mint 18.2 Sonya
    Release:	18.2
    Codename:	sonya

Nevertheless, their packaged lifehackrf is rather old:
    libhackrf-dev  2015.07.2-6

But with your action(s) it now worked: git pull; configure; make and make suidinstall did compile and install. Thank you very much.

Detlef Bloch