kismetwireless/kismet #11: Ampersands not escaped in <wps-manuf>

kurankat, 2017-09-13:
Version: 2016_07_R1-1
Platform: Arch Linux ARM (Raspberry Pi 3)

I've come across some instances in netxml output where the content is not properly escaped, and cannot be used by a number of XML parsing tools.
While <essid> and <client-manuf> seem to contain only escaped content, the <wps-manuf> does not always do so. 
I have attached  an excerpt from a .netxml file showing both properly escaped ampersands in some fields, and an unescaped one in <wps-manuf>.
[unescaped_XML_content.txt](https://github.com/kismetwireless/kismet/files/1298751/unescaped_XML_content.txt)

kismetwireless, 2017-09-13:
Ah; good catch.  I'll take a quick look, but the netxml format is almost certainly going to die in favor of a new more flexible log in the very near future...  I'll see about fixing it in git, but that code will likely not see another release; the new system will supplant it

kismetwireless, 2017-09-13:
Yep - you were totally right.  XML escaping wasn't being done on the WPS manufacturer, just the device manufacturer itself.

Fixed in commit f6708d6bdaad89702b8d845ff9c5da792b129e47 in the **kismet-stable** branch (you'll need to make sure to check out that branch not master).

Thanks!

closed 2017-09-13 by kismetwireless