kismetwireless/kismet #13: Fails on make

JoviDeCroock, 2017-10-25:
my ./configure said everything was fine but when i'm compiling the newest version of this repository i get this error:

[image]

kismetwireless, 2017-10-25:
Looks like you have a weird version of libmicrohttpd; what version are you trying to compile against?  0.9.30 is known to work.

JoviDeCroock, 2017-10-25:
I have version 0.9.19

kismetwireless, 2017-10-25:
It's possible you've still got an old copy of microhttpd kicking around, I can't replicate & haven't heard of this error happening before, unforuntately...

JoviDeCroock, 2017-10-25:
I fixed it with:

```
cd
wget ftp://ftp.gnu.org/gnu/libmicrohttpd/libmicrohttpd-0.9.30.tar.gz
tar zxvf libmicrohttpd-0.9.30.tar.gz
mv libmicrohttpd-0.9.30 libmicrohttpd
cd libmicrohttpd
./configure
make
make install
```

want me to make a PR to add this in the readme? :) 
Thanks for the quick responses btw, love your work

kismetwireless, 2017-10-26:
I just rewrote that part of the code to avoid using quiesce entirely, now it disables throwing errors in the panic handler and just kills the mhd daemon.
That, hopefully, fixes it for any version.  I'm going to close this ticket for now but re-open it or toss a new one in if it doesn't seem to solve it!

closed 2017-10-26 by kismetwireless