kismetwireless/kismet #16: Possible bug in Kismet Web UI GPS

bcavileer, 2017-11-11:
I'm using Kismet with the web UI, version
commit 34c9da5a037934bf3b77525b9cdc410553c3c1c0
Author: Mike Kershaw / Dragorn <dragorn@kismetwireless.net>
Date:   Tue Nov 7 17:21:49 2017 -0500

I'm not seeing GPS in the status bar, "No GPS Available" "GPS Connection Lost..."
[image]

I've enabled GPS in /usr/local/etc/kismet.conf
```
# New GPS configuration
# gps=type:options
#
# gps=serial:device=/dev/ttyACM0,name=laptop
gps=gpsd:host=localhost,port=2947
# gps=virtual:lat=123.45,lon=45.678,alt=1234
```

I'm running gpsd and gpsmon.  Everything is looking good there.

I'm seeing success messages in the Kismet server logs
`INFO: GPSGpsdV2 connecting to GPSD server on localhost:2947`
`INFO: GPSGpsdV2 connected to a JSON-enabled GPSD version 3.15, turning on JSON mode`

I'm seeing the web UI poll for location.json and getting a response of
```
{  
   "kismet.common.location.lat":38.530914,
   "kismet.common.location.lon":-73.643761,
   "kismet.common.location.alt":7.000000,
   "kismet.common.location.speed":0.000000,
   "kismet.common.location.heading":0.000000,
   "kismet.common.location.fix":2,
   "kismet.common.location.valid":1,
   "kismet.common.location.time_sec":1510359306,
   "kismet.common.location.time_usec":142349,
   "kismet.common.location.gps_uuid":"17180636-0000-0000-0000-0000e7e30ae8"
}
```

I'm not sure if this is a bug or I'm missing something obvious?  Should the Web UI display GPS info in the status bar?

bcavileer, 2017-11-11:
It seems I was getting a bad GPS fix. This is working fine now!

closed 2017-11-11 by bcavileer

XD346, 2017-12-24:
I have had a GPS problem as well.  
The kismet.conf file not sure if that has any thing to do with it.
The only thing I had changed was gpsdevice=/dev/ttyUSB0 . 
But nothing seems to work. BR-355s4 GPS