kismetwireless/kismet #17: Failed to open user plugin directory

jlash1314, 2017-11-25:
I'm getting an error when I open kismet. I'm using the git master version that I downloaded and compiled yesterday. It gives me "Failed to open user plugin directory (/root/.kismet//plugins/): No such file or directory". I noticed that there is an extra forward slash is the directory tree between /.kismet and /plugins/. My linux is limited and rusty so please correct me if I'm wrong. I'm not really sure where the extra / is coming from. 
JT

kismetwireless, 2017-11-25:
Slashes are fine in a unix path; multiple slashes are treated as one.

closed 2017-11-25 by kismetwireless

kismetwireless, 2017-11-25:
BTW i've changed the message to be flagged as info instead of error and slightly changed the message as well to be less alarming.