kismetwireless/kismet #21: Problem kismet as a service

aimdev, 2017-12-20:
I have compiled kismet from a git clone today, 20-dec-2017 on
a Debian Jessie uptodate system I observerd the following.

kismet when run from the command line works ok
kismet -c wlan0

as does 
kismet 
where the line source=wlan0 has been uncommented in /usr/local/etc/kismet.conf

However, after following the instructions to install kismet as a service,
enabling the service, when starting and checking the status, the following is observerd

● kismet.service - Kismet
   Loaded: loaded (/etc/systemd/system/kismet.service; enabled)
   Active: failed (Result: start-limit) since Wed 2017-12-20 16:47:12 GMT; 13s ago
  Process: 810 ExecStart=/usr/local/bin/kismet (code=exited, status=217/USER)
 Main PID: 810 (code=exited, status=217/USER)

Dec 20 16:47:12 debian systemd[1]: Unit kismet.service entered failed state.
Dec 20 16:47:12 debian systemd[1]: kismet.service holdoff time over, scheduling restart.
Dec 20 16:47:12 debian systemd[1]: Stopping Kismet...
Dec 20 16:47:12 debian systemd[1]: Starting Kismet...
Dec 20 16:47:12 debian systemd[1]: kismet.service start request repeated too quickly, refusing to start.
Dec 20 16:47:12 debian systemd[1]: Failed to start Kismet.
Dec 20 16:47:12 debian systemd[1]: Unit kismet.service entered failed state.

rebooting does not resolve the problem.

Please can you asist me in fixing this problem.

Thanks

kismetwireless, 2017-12-20:
Likely you need to create a user or alter the systemd scripts - they're definitely not ready/designed for deployment without tinkering.  I don't see anything in that log which helps diagnose, sorry.

closed 2017-12-20 by kismetwireless

kismetwireless, 2017-12-21:
I've revamped the systemd service file to run as root and included directions in the readme in the systemd packaging dir on how to configure it to a specific user

aimdev, 2017-12-21:
Hi
Sort of works, using my modified file, but cannot stop it with service kismet stop, and service kismet status always reports a failure, but it is running as the web site proves by showing received data. Also works on reboot.  The delays may be allowing for the interfaces to come up.
The libnm (the NetworkManager control library) was not found message is a pain as the version of libnm does not appear to be detected during configure, and the later version of this library is not readily available on debian jessie, perhaps I need to build it later, as a debian upgrade is not possible at the moment.

[Unit]
Description=Kismet
ConditionPathExists=/usr/local/bin/kismet
After=network.target auditd.service

[Service]
User=root
Group=root
Type=simple
ExecStart=/usr/local/bin/kismet
KillMode=process
TimeoutStartSec=10
TimeoutStopSec=0
SysVStartPriority=99
Restart=on-failure
RestartSec=30

[Install]
WantedBy=multi-user.target