kismetwireless/kismet #22: Building kismet from source fails with `configure: error: libz is required and could not be found`

dbarella, 2017-12-23:
I'm trying to build kismet from source on the most recent stable version of kubuntu, AKA ubuntu with KDE:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.3 LTS
Release:        16.04
Codename:       xenial
```

I made sure to run the recommended package installs already, and I do have (an instance of) zlib installed already per the ubuntu defaults:

```
$ apt list zlib1g -a
Listing... Done
zlib1g/xenial-updates,now 1:1.2.8.dfsg-2ubuntu4.1 amd64 [installed]
zlib1g/xenial 1:1.2.8.dfsg-2ubuntu4 amd64
```

However, it looks like the exact command as installed under ubuntu is not exactly what `./configure` expects:

```
$ deflate
deflate: command not found

$ zlib-flate --help
Usage: zlib-flate { -uncompress | -compress }
```

And it looks like `configure` is checking for `deflate` on the command line, not for the zlib headers:

```
$ ./configure
... stuff ...
checking for deflate in -lz... no
configure: error: libz is required and could not be found
```

I think the issue is arising from [this commit](https://github.com/kismetwireless/kismet/commit/4a702b5b18755a3e8af9be96a3fc0ee9dcc65fd1#diff-e2d5a00791bce9a01f99bc6fd613a39d) -- I'm not able to point out exactly what's going on because there's a lot of indirection in that config script and I haven't entirely understood the check that it's running.

---

The questions I have are these:

- Is there a particular package I should install to convince `configure` that zlib is installed on my system?
- Is there a point to modifying `configure` so that it can pick up `zlib-flate` instead?

I also wonder if aliasing `deflate` to `zlib-flate -uncompress` would be "good enough", though I suspect the flags won't line up with what kismet is expecting.

kismetwireless, 2017-12-23:
Unfortunately, I have no idea what kubuntu has done differently - on stock ubuntu (at least from 14 through 17) the list of packages in the readme gets everything configure needs.

Autoconf looks for the deflate function inside libz:

```
# We need libz
AC_CHECK_LIB([z], [deflate],
    ¦   ¦   ¦usez=1,
    ¦   ¦   ¦AC_MSG_ERROR([libz is required and could not be found]))
LIBS="$LIBS -lz"
```
so it expects libz.so to contain the deflate command; nothing in the kismet config should be calling zlib-deflate as a command as far as i can see... that turns into:

```# We need libz
{ $as_echo "$as_me:${as_lineno-$LINENO}: checking for deflate in -lz" >&5
$as_echo_n "checking for deflate in -lz... " >&6; }
if ${ac_cv_lib_z_deflate+:} false; then :
  $as_echo_n "(cached) " >&6
else
  ac_check_lib_save_LIBS=$LIBS
LIBS="-lz  $LIBS"
cat confdefs.h - <<_ACEOF >conftest.$ac_ext
/* end confdefs.h.  */

/* Override any GCC internal prototype to avoid an error.
   Use char because int might match the return type of a GCC
   builtin and then its argument prototype would still apply.  */
#ifdef __cplusplus
extern "C"
#endif
char deflate ();
int
main ()
{
return deflate ();
  ;
  return 0;
}
_ACEOF
if ac_fn_cxx_try_link "$LINENO"; then :
  ac_cv_lib_z_deflate=yes
else
  ac_cv_lib_z_deflate=no
fi
rm -f core conftest.err conftest.$ac_objext \
    conftest$ac_exeext conftest.$ac_ext
LIBS=$ac_check_lib_save_LIBS
fi
{ $as_echo "$as_me:${as_lineno-$LINENO}: result: $ac_cv_lib_z_deflate" >&5
$as_echo "$ac_cv_lib_z_deflate" >&6; }
if test "x$ac_cv_lib_z_deflate" = xyes; then :
  usez=1
else
  as_fn_error $? "libz is required and could not be found" "$LINENO" 5
fi

LIBS="$LIBS -lz"
```

so it should work so long as you have libz.so; it's not even using the libz.h headers...

dbarella, 2017-12-23:
Yeah, that's about as much as I understood from starting at `config` for a bit. I'm going to chalk this up to a package install problem -- as is the (frustrating) norm, I resolved the issue by doing some package reinstallation and rebooting:

```
$ sudo apt update
<snip>
$ sudo apt upgrade
<snip>
$ sudo apt autoremove
<snip>

# Note: I intentionally didn't force a reinstall;
# that would have been the next step
# if things still weren't working after a reboot
$ sudo apt install \
  build-essential 
  libmicrohttpd-dev \
  git \
  libnl-3-dev \
  libnl-genl-3-dev \
  libcap-dev \
  libpcap-dev \
  libncurses5-dev \
  libnm-dev \
  libdw-dev \
  libsqlite3-dev 
<build-essential and git are ignored, everything else is already installed>

$ reboot
$ ./configure
<Problems Magically Solved!>
```

After this, running `configure` proceeded as normal, which is too bad because I wanted to understand what was going wrong. Oh well. To future readers who use kubuntu, try these steps and see if it works for you! Make sure to check the package list against the `README` first, though.

dbarella, 2017-12-23:
p.s. thanks a bunch for the _really_ swift response, @kismetwireless!

closed 2017-12-23 by dbarella

kismetwireless, 2017-12-23:
Interesting - thanks for the followup!

I'd suggest, perhaps, a 'sudo ldconfig' in the future, too - that can help linked libs get set up properly in the system.

In a perfect world, the package manager does this - it's usually only an issue when compiling from source, but there's a chance that could straighten it out.

Either way, glad it's working now!