kismetwireless/kismet #24: Supported wifi cards for DroneID?

techspy2000, 2017-12-31:
I am unable to detect any DJI drones with the latest Kismet (Mavic or Spark). I understand there may be an issue detecting the Spark when using the RC (which I am) but I get nothing from the Mavic either (FW 3.000). I am running on an older laptop/wifi card and will try swapping to a different one but am curious if there are specific ones that are know to work or not work?

kismetwireless, 2018-01-02:
Droneid is an IE tag inside a beacon; as such basically any card capable of monitor mode will be able to see it.  Beacons are sent at the slowest, simplest data rates which means you ought to be able to see them without worrying about multistream, 11n/11ac encoding, firmware bugs causing bad data reporting from your card, etc - they're (quite likely deliberately) in the easiest frames to capture so if you're seeing devices and access points in Kismet at all, your problem most likely isn't that you're not capturing.

I don't like to recommend specific cards as a matter of principle because vendors constantly change the chipsets in cards without changing the model, making it impossible to distinguish good, or even workable, cards from useless cards.

In general, anything ath9k based is generally good (mostly minipci-e, a few USB cards, but it's hard to find the exact model which has the right chipset).

DroneID frames are only sent from DJI July firmware and newer, and as of their newest firmware and DJI Go app, only when enabled in the app (previously it was silently enabled on all devices).  They are only sent in Wi-Fi mode, they appear to not be sent when using the RC (or if they are, they're a modified format I don't know yet), and they're only sent when the props are on and the drone is in the air, if I recall.  You'll need to be outside and have the drone in the air.

You can look at your pcaps for beacons with IE 221 with an OUI / vendor specific category of 26:37:12 ; they'll appear in wireshark, etc.  You can take a look for them in your pcappi logs (or running wireshark at the same time as kismet); there are two formats of the tag (action and flight record); kismet only decodes the action/telemetry currently; the flight record format doesnt' contain too much useful data but support for it will be coming sooner or later.

Hope that helps

closed 2018-01-02 by kismetwireless

kismetwireless, 2018-01-03:
Also it looks like the latest DJI firmware has done *something* to droneid - either deliberately or accidentally, it is now very broken.  Beacons are sent with the IE tag included, but the content of the IE tag is zeroed; either they have zeroed it out deliberately, or they have broken their droneid population code in current firmware; I haven't seen the data appear anywhere else in packets.