kismetwireless/kismet #27: Remote GPS

D4rkC00d3r, 2018-01-10:
I have Kismet working perfect with remote packet capture over reverse ssh and port forwarding. Is is possible to do the same with the GPS attached to the remote device. I have been trying all day with the settings and GPS is working fine on the remote device I just can't seem to get it showing on the rear device without it plugged into the rear?

kismetwireless, 2018-01-10:
It's on the short list of things to do; the protocol has a spot for it, it's just not currently populated.

Most likely the direct gps support will be for sending a fixed location; a separate tool will have to be written to do full live gps on a remote node; because of how the remote capture works in general, I'd like to avoid parsing NMEA in a root daemon, but it should be possible to connect a remote gps and bind it to a remote source (or sources).

I'll leave the issue open, it might be a little bit before remote gps gets done but it's definitely on the list.

mo8Zomo0, 2018-03-02:
Hi!

I have an issue which might be related sort of or be a solution to this issue (I might also not have understood his issue).

Should'nt GPSD an ok way to handle this? Thats at least what I wanted to use even only localhost on my Macbook Air (Sierra). At least the GPSD info should be easy tunneled. But ok, maybe this case
is about that hooking that up to some remote instance is not possible. 

Anyhow, let me know if I shall open a new ticket or if its related:

I have a usb ublox-7 stick and that works fine if I use the 
`gps=serial:device=/dev/cu.usbmodemfa131,name=laptop`

> _INFO: GPSSerial defaulting to 4800 baud for GPS device, specify baud= 
>       option if your device uses a different speed._

And I can see it working on the webinterface.

Now using gpsd config:
gps=gpsd:host=localhost,port=2947

When I try to start it up with GPSd the kismet server startup hangs here:

> _INFO: GPSGpsdV2 connected to a JSON-enabled GPSD version 3.17, turning on
>       JSON mode
> FATAL - Capture source did not get PING from Kismet for over 15 seconds; shutting down
> FATAL - Capture source did not get PING from Kismet for over 15 seconds; shutting down_
> 

I did a tcpdump on lo0 and I can see that there is some communication, though I wonder if I should see some clear text command to enable logging on the gpsd:

_$ sudo tcpdump -nl -s0 -A -i lo0 port 2947_ 

> tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
> listening on lo0, link-type NULL (BSD loopback), capture size 262144 bytes
> 20:08:50.990515 IP 127.0.0.1.59819 > 127.0.0.1.2947: Flags [S], seq 1587294081, win 65535, options [mss 16344,nop,wscale 5,nop,nop,TS val 541747035 ecr 0,sackOK,eol], length 0
> E..@{.@.@...............^./..........4....?........
>  Jg[........
> 20:08:50.990626 IP 127.0.0.1.2947 > 127.0.0.1.59819: Flags [S.], seq 2555905970, ack 1587294082, win 65535, options [mss 16344,nop,wscale 5,nop,nop,TS val 541747035 ecr 541747035,sackOK,eol], length 0
> E..@k.@.@................X..^./......4....?........
>  Jg[ Jg[....
> 20:08:50.990640 IP 127.0.0.1.59819 > 127.0.0.1.2947: Flags [.], ack 1, win 12759, options [nop,nop,TS val 541747035 ecr 541747035], length 0
> E..4.X@.@...............^./..X....1..(.....
>  Jg[ Jg[
> 20:08:50.990653 IP 127.0.0.1.2947 > 127.0.0.1.59819: Flags [.], ack 1, win 12759, options [nop,nop,TS val 541747035 ecr 541747035], length 0
> E..4..@.@................X..^./...1..(.....
>  Jg[ Jg[
> 20:08:50.990748 IP 127.0.0.1.2947 > 127.0.0.1.59819: Flags [P.], seq 1:85, ack 1, win 12759, options [nop,nop,TS val 541747035 ecr 541747035], length 84
> E....S@.@................X..^./...1..|.....
>  Jg[ Jg[{"class":"VERSION","release":"3.17","rev":"3.17","proto_major":3,"proto_minor":12}
> 
> 20:08:50.990780 IP 127.0.0.1.59819 > 127.0.0.1.2947: Flags [.], ack 85, win 12756, options [nop,nop,TS val 541747035 ecr 541747035], length 0
> E..4.N@.@...............^./..X....1..(.....
>  Jg[ Jg[

$ date; git status
Fri Mar  2 20:16:07 CET 2018
On branch master
Your branch is up-to-date with 'origin/master'.

$ kismet -v --debug
Kismet 2017-00-DEV

$ gpsd -V
gpsd: 3.17 (revision 3.17)

kismetwireless, 2018-03-02:
Unrelated, if you can throw this into another issue I'll follow up there though.

XD346, 2018-03-19:
But I have Ubuntu 17.10
The kismet version worked with my GPSD and now that I have upgraded to the unstable version. The web GUI GPS fail to connect with editing the kismet.con file dev/ttyUSB  
I am running Ubuntu 17.10 my gps is BR-355s4  tried to get the debug file but not that experience .

kismetwireless, 2018-03-19:
I'm going to close this bug since it keeps collecting unrelated things; @D4rkC00d3r it's still on the shortlist; the capture protocol is changing soon and will accommodate this I hope.

kismetwireless, 2018-04-23:
@D4rkC00d3r finally got to this; you can do a fixed gps on a remote source now with --fixed-gps=[lat,lon,alt] or --fixed-gps=[lat,lon]; I'm not sure live GPS on a remote source makes a ton of sense since a remote source kind of by definition needs a high-bandwidth connection which would make it non-mobile.

I've got some ideas for splitting out a remote gps-only interface so that the root code can stay slim and single-purpose, but it's further down the list.

closed 2018-04-23 by kismetwireless