kismetwireless/kismet #30: kismet only reports WEP encryption for WPA capable wifi network

pcabreracamara, 2018-01-19:
Hi all,

I realize that for some wifi networks, kismet only reports "WEP" crypt_set (in the web UI and JSON API). For example;

```
"kismet.device.base.name": "GABRIELA",
"kismet.device.base.username": "",
"kismet.device.base.type": "Wi-Fi AP",
"kismet.device.base.basic_type_set": 1,
"kismet.device.base.crypt": "",
"kismet.device.base.basic_crypt_set": 2,
```

However, using airodump and capturing to PCAP file, the beacon frame only sends Specific Vendor MS WPA information element, but is not sending "RSN Information" element:

Beacon frame in wireshark:
[image]

However airodump mark encryption as WPA:
[image]

Is there any way to force kismet to look/analyze "Specific Vendor MS WPA information element" as airodump does?

Regards,
Pedro

kismetwireless, 2018-01-22:
Kismet does analyze that field; if you're able to provide a pcap where kismet decodes it wrong I'll take a deeper look.

pcabreracamara, 2018-01-24:
Hi all,

Find here a pcap with the beacon probes of "RESTAURANTE_CANA" wifi network, shown WEP by kissmet in the Web UI, JSON and SQLite DB:

pcap file:
[all_sources_CANA.pcapng.zip](https://github.com/kismetwireless/kismet/files/1660553/all_sources_CANA.pcapng.zip)

JSON file:
[device_CANA.json.zip](https://github.com/kismetwireless/kismet/files/1660558/device_CANA.json.zip)

Web UI:
[image]

Kismet sqlite3 db:
[image]

I check this behaviour with at least 3 more wifi networks (all of them WPA/CCMP/PSK). LEt me know anything else you need from me.
 
Thank you for your response !!

kismetwireless, 2018-01-24:
Ah, I see now - you're completely right; kismet parses the WPA fields, and the RSN field, but did not parse the WFA-WPA 221 field.  I'm working on a decoder now & will update here; thanks for the pcap!

kismetwireless, 2018-01-24:
As of kismet commit 4b3fc175783828f5ef7bb1926e750b76e58d7d1f it should be parsing wfa/wpa tags now and incorporating them into the encryption info; it parses the pcap you sent along.  I'll close this for now but if you see more problems feel free to re-open it!

Thanks for the heads up and the pcap

closed 2018-01-24 by kismetwireless