kismetwireless/kismet #32: KismetRest - problems specifying fields to return from smart_device_list

gyre, 2018-02-09:
I'm having problems when I attempt to restrict which fields are returned from the kismet REST server.

I'm building from the master GIT branch of kismet, running the server on a raspberry PI 3.

```
#!/usr/bin/env python

import time
import KismetRest
import argparse

def per_device(d):
    # print d
    # for k in d.keys():
    #     print k

    if d["kismet.device.base.type"] != "Wi-Fi AP":
        print "%s %.2fs ago" % (d['kismet.device.base.macaddr'], (
            time.time() - d['kismet.device.base.last_time']))

uri = "http://localhost:2501"

parser = argparse.ArgumentParser(description='Kismet demo code')

parser.add_argument('--uri', action="store", dest="uri")

results = parser.parse_args()

if results.uri is not None:
    uri = results.uri

kr = KismetRest.KismetConnector(uri)

fields = [
    "kismet.device.base.type", "kismet.device.base.macaddr",
    "kismet.device.base.last_time"
]

# kr.smart_device_list(ts=-60, callback=per_device)
kr.smart_device_list(fields=fields, callback=per_device)

```

This is a wireshark compilation of the conversation between my code and the kismet server.

```
POST /devices/last-time/0/devices.ekjson HTTP/1.1
Host: mome:2501
Content-Length: 136
Accept-Encoding: gzip, deflate
Accept: */*
User-Agent: python-requests/2.9.1
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded

json=%7B%22fields%22%3A+%5B%22kismet.device.base.type%22%2C+%22kismet.device.base.macaddr%22%2C+%22kismet.device.base.last_time%22%5D%7DHTTP/1.1 200 OK
Connection: Keep-Alive
Transfer-Encoding: chunked
Expires: Sat, 01 Jan 2000 00:00:00 GMT
Pragma: no-cache
Cache-Control: no-cache
Access-Control-Allow-Origin: *
Content-Type: application/json
Last-Modified: Fri, 09 Feb 2018 15:49:52 GMT
Date: Fri, 09 Feb 2018 15:49:52 GMT

7C
Invalid request: [json.exception.parse_error.101] parse error at 11: syntax error - invalid literal; last read: '"fields":+'
0
```

This is the python backtrace:

```
Failed to parse JSON: http://mome:2501/devices/last-time/0/devices.ekjson, No JSON object could be decoded, Invalid request: [json.exception.parse_error.101] parse error at 10: syntax error - invalid literal; last read: '"regex":+'
Traceback (most recent call last):
  File "regex_device_oui.py", line 66, in <module>
    kr.smart_device_list(callback = per_device, regex = regex)
  File "/home/ange/.local/lib/python2.7/site-packages/KismetRest/__init__.py", line 521, in smart_device_list
    (r, v) = self.__post_json_url("devices/last-time/{}/devices.ekjson".format(ts), cmd, callback, cbargs, stream = True)
  File "/home/ange/.local/lib/python2.7/site-packages/KismetRest/__init__.py", line 359, in __post_json_url
    return (r.status_code, self.__process_json_stream(r, callback, cbargs))
  File "/home/ange/.local/lib/python2.7/site-packages/KismetRest/__init__.py", line 233, in __process_json_stream
    r = self.__process_json_object(r, line, callback, args)
  File "/home/ange/.local/lib/python2.7/site-packages/KismetRest/__init__.py", line 203, in __process_json_object
    raise KismetRequestException("Unable to parse JSON on req {}: {}".format(r.url, e.message), r.status_code)
KismetRest.KismetRequestException: Unable to parse JSON on req http://mome:2501/devices/last-time/0/devices.ekjson: No JSON object could be decoded
```

If I can help debug this further... please shout.

-- gyre --

kismetwireless, 2018-02-09:
Your exact code runs fine for me (copied and pasted from your post)

dragorn@drd1812-25:~/src$ python ./fields.py --uri http://kismet:2501
D4:63:C6:DB:56:84 16300.43s ago
1E:03:81:E7:C8:D9 139658.43s ago
F6:EC:3E:ED:F0:54 139627.43s ago
53:C2:1B:61:A7:1C 69379.43s ago
7D:4B:9E:AE:2E:D9 69325.43s ago
5D:DB:F9:D2:12:E2 69256.43s ago
58:36:DF:24:C9:CD 69127.43s ago
60:B5:1C:7B:9F:EB 69063.43s ago
6A:24:B0:D1:1A:50 68908.43s ago

Are you sure you're running the latest KismetRest?  Try re-installing the python library...

kismetwireless, 2018-02-09:
This also works for me:

curl --data 'json=%7B%22fields%22%3A+%5B%22kismet.device.base.type%22%2C+%22kismet.device.base.macaddr%22%2C+%22kismet.device.base.last_time%22%5D%7D' http://kismet:2501/devices/last-time/0/devices.ekjson

which would lead me to suspect that you've got some wacky data in your kismet devices which isn't being serialized correctly, but that doesn't match the error you're showing.

What distribution on the pi?

gyre, 2018-02-09:
I tried your curl example:
```
curl --data 'json=%7B%22fields%22%3A+%5B%22kismet.device.base.type%22%2C+%22kismet.device.base.macaddr%22%2C+%22kismet.device.base.last_time%22%5D%7D' http://mome:2501/devices/last-time/0/devices.ekjson
```
which resulted with:
```
Invalid request: [json.exception.parse_error.101] parse error at 11: syntax error - invalid literal; last read: '"fields":+'
```
I've just deleted kismet completely from the pi and re-cloned from github.  building now.  However, this was cloned last night, so I'm not expecting miracles.

```
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

kismetwireless, 2018-02-09:
All I can think is that somehow the json parser functions differently on your compiler; I can't recreate the behavior on any of several ubuntu flavors going back to 14.04...

kismetwireless, 2018-02-09:
Alright - latest git code rolls off the nlohmann json.hpp for C++11 and onto an older API in jsoncpp; I also had trouble with the c++11 code on openwrt, so weird parser errors are sort of the straw here.

You'll need to do a ./configure and a rm *.d.

"works for me", but the old code did, too, so I'd be interested to see how it performs on your system.

kismetwireless, 2018-02-14:
Looks like it's linked to an extremely old version of libmicrohttpd in debian jessie; you'll need to grab a newer version; 1.9.30 or newer at least.

Old microhttpd doesn't parse the urlencoded data correctly and passes raw '+' characters.

gyre, 2018-02-14:
Cool.  Will try that today.  Thanks!! :)

gyre, 2018-02-14:
Yup, you've nailed it.  backporting libmicrohttpd(-dev) 0.9.51 from debian stretch seems to work fine.  :)

kismetwireless, 2018-02-14:
Awesome; I've added a check to autoconf which will bail if it finds this version of microhttpd, as well.  Thanks for the heads up!

closed 2018-02-14 by kismetwireless

tomsaul, 2019-04-25:
For those trying to test from a Windows (10) system, the parameter must in in quotes

curl --data "json=%7B%22fields%22%3A+%5B%22kismet.device.base.type%22%2C+%22kismet.device.base.macaddr%22%2C+%22kismet.device.base.last_time%22%5D%7D" http://x:2501/devices/last-time/0/devices.ekjson

Also note that the 2019-04 Raspbian release works out-of-the-box