kismetwireless/kismet #33: Kismet crashing while scanning all channels

djjerdog, 2018-02-11:
I just updated to the lastest code and I'm getting this error after a minute our so when scanning all channels.

Kismet server terminated.  Last output:
4    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.22, at 0xffff8143d347, in std::__throw_ios_failure(char const*)
#3    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.22, at 0xffff8141501b, in __cxa_throw
#2    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.22, at 0xffff81414d97, in std::terminate()
#1    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.22, at 0xffff81414d4b, in 
#0    Source "/home/wlanpi/kismet/kismet_server.cc", line 417, in TerminationHandler
        415: #ifndef DISABLE_BACKWARD
        416:     using namespace backward;
      > 417:     StackTrace st; st.load_here(32);
        418:     Printer p; p.print(st);
        419: #endif
Kismet exited.

kismetwireless, 2018-02-11:
Ok; if you could take a look at the debugging section in the readme and post the stack trace I'll take a look, thanks!

kismetwireless, 2018-02-11:
Coordinated via email; should be fixed in the latest commits (missing try/catch on rmm IE fields)

closed 2018-02-11 by kismetwireless