kismetwireless/kismet #38: GPSd issue

mo8Zomo0, 2018-03-14:
Hi!

I am using a Macbook Air 2011 with Sierra 10.12.6
with a usb ublox-7 stick and that works fine if I use the
gps=serial:device=/dev/cu.usbmodemfa131,name=laptop

INFO: GPSSerial defaulting to 4800 baud for GPS device, specify baud=
option if your device uses a different speed.

And I can see it working on the webinterface.

Now using gpsd config:
gps=gpsd:host=localhost,port=2947

When I try to start it up with GPSd the kismet server startup hangs here:

INFO: GPSGpsdV2 connected to a JSON-enabled GPSD version 3.17, turning on
JSON mode
FATAL - Capture source did not get PING from Kismet for over 15 seconds; shutting down
FATAL - Capture source did not get PING from Kismet for over 15 seconds; shutting down

I did a tcpdump on lo0 and I can see that there is some communication, though I wonder if I should see some clear text command to enable logging on the gpsd:

$ sudo tcpdump -nl -s0 -A -i lo0 port 2947

tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on lo0, link-type NULL (BSD loopback), capture size 262144 bytes
20:08:50.990515 IP 127.0.0.1.59819 > 127.0.0.1.2947: Flags [S], seq 1587294081, win 65535, options [mss 16344,nop,wscale 5,nop,nop,TS val 541747035 ecr 0,sackOK,eol], length 0
E..@{.@.@...............^./..........4....?........
Jg[........
20:08:50.990626 IP 127.0.0.1.2947 > 127.0.0.1.59819: Flags [S.], seq 2555905970, ack 1587294082, win 65535, options [mss 16344,nop,wscale 5,nop,nop,TS val 541747035 ecr 541747035,sackOK,eol], length 0
E..@k.@.@................X..^./......4....?........
Jg[ Jg[....
20:08:50.990640 IP 127.0.0.1.59819 > 127.0.0.1.2947: Flags [.], ack 1, win 12759, options [nop,nop,TS val 541747035 ecr 541747035], length 0
E..4.X@.@...............^./..X....1..(.....
Jg[ Jg[
20:08:50.990653 IP 127.0.0.1.2947 > 127.0.0.1.59819: Flags [.], ack 1, win 12759, options [nop,nop,TS val 541747035 ecr 541747035], length 0
E..4..@.@................X..^./...1..(.....
Jg[ Jg[
20:08:50.990748 IP 127.0.0.1.2947 > 127.0.0.1.59819: Flags [P.], seq 1:85, ack 1, win 12759, options [nop,nop,TS val 541747035 ecr 541747035], length 84
E....S@.@................X..^./...1..|.....
Jg[ Jg[{"class":"VERSION","release":"3.17","rev":"3.17","proto_major":3,"proto_minor":12}

20:08:50.990780 IP 127.0.0.1.59819 > 127.0.0.1.2947: Flags [.], ack 85, win 12756, options [nop,nop,TS val 541747035 ecr 541747035], length 0
E..4.N@.@...............^./..X....1..(.....
Jg[ Jg[

$ date; git status
Fri Mar 2 20:16:07 CET 2018
On branch master
Your branch is up-to-date with 'origin/master'.

$ kismet -v --debug
Kismet 2017-00-DEV

$ gpsd -V
gpsd: 3.17 (revision 3.17)

kismetwireless, 2018-03-14:
This should be fixed as of commit...  ee3a0b67c1ab6789db0c46a0e7d3c567029e279e .  It turns out that OSX doesn't implement non-blocking read() on a nonblocking socket; it performs a blocking operation.

I respun the code to use recv and send instead of read and write for all socket ops and that seemed to address the timeout problems on OSX; give that commit (or newer) a shot.

mo8Zomo0, 2018-03-14:
Yes, I am now on d032c652 and I can confirm that accessing the gpsd does not cause it to hang anymore and seems to work.

closed 2018-03-14 by mo8Zomo0