kismetwireless/kismet #39: protobuf error

mo8Zomo0, 2018-03-14:
Hi! 

I just tried to upgrade my kismet with the master from git, but it failed compiling (MacOS 10.12.6).
Which version of proto is needed? I tried all which were available in homebrew but it always fails.
Last time it was like 2 weeks ago I tried it, there compiling was no issue. 

Thats how I compile:
`$ make clean ; CFLAGS="-I/opt/local/include" LDFLAGS="-L/usr/local/opt/protobuf@3.1/lib -L/opt/local/lib" CPPFLAGS="-I/usr/local/opt/protobuf@3.1/include -I/opt/local/include" ./configure && make all&& sudo make install
`

and thats the current error:
`g++ -std=gnu++11 -MM -Wall -g -I. -g -O2  -I/usr/local/opt/protobuf@3.1/include -I/opt/local/include -I./msgpack-1.4.1/include  protobuf_cpp/kismet.pb.cc | sed -e "s/\.o/\.cc.o/" > protobuf_cpp/kismet.pb.cc.d
In file included from protobuf_cpp/kismet.pb.cc:5:
protobuf_cpp/kismet.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
protobuf_cpp/kismet.pb.h:18:2: error: incompatible with your Protocol Buffer headers. Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
protobuf_cpp/kismet.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
3 errors generated.
g++ -std=gnu++11 -Wall -g -I. -g -O2  -I/usr/local/opt/protobuf@3.1/include -I/opt/local/include -I./msgpack-1.4.1/include  -c protobuf_cpp/kismet.pb.cc -o protobuf_cpp/kismet.pb.cc.o
In file included from protobuf_cpp/kismet.pb.cc:5:
protobuf_cpp/kismet.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
protobuf_cpp/kismet.pb.h:18:2: error: incompatible with your Protocol Buffer headers. Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
protobuf_cpp/kismet.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
In file included from protobuf_cpp/kismet.pb.cc:5:
In file included from protobuf_cpp/kismet.pb.h:22:
In file included from /usr/local/opt/protobuf@3.1/include/google/protobuf/arena.h:54:
In file included from /usr/local/opt/protobuf@3.1/include/google/protobuf/stubs/atomic_sequence_num.h:33:
In file included from /usr/local/opt/protobuf@3.1/include/google/protobuf/stubs/atomicops.h:199:
/usr/local/opt/protobuf@3.1/include/google/protobuf/stubs/atomicops_internals_macosx.h:47:9: warning: 
      'OSAtomicCompareAndSwap32' is deprecated: first deprecated in macOS 10.12 - Use
      std::atomic_compare_exchange_strong_explicit(std::memory_order_relaxed) from <atomic> instead
      [-Wdeprecated-declarations]
    if (OSAtomicCompareAndSwap32(old_value, new_value,
        ^
/usr/include/libkern/OSAtomicDeprecated.h:502:9: note: 'OSAtomicCompareAndSwap32' has been explicitly marked deprecated
      here
bool    OSAtomicCompareAndSwap32( int32_t __oldValue, int32_t __newValue, volatile int32_t *__theValue );
        ^
In file included from protobuf_cpp/kismet.pb.cc:5:
In file included from protobuf_cpp/kismet.pb.h:22:
`

`$ brew info protobuf
protobuf: stable 3.5.1 (bottled), HEAD
Protocol buffers (Google's data interchange format)
https://github.com/google/protobuf/
/usr/local/Cellar/protobuf/3.2.0_1 (256 files, 15.9MB)
  Poured from bottle on 2017-04-25 at 17:16:11
/usr/local/Cellar/protobuf/3.5.1_1 (267 files, 18.5MB) *
  Poured from bottle on 2018-03-14 at 17:20:08
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/protobuf.rb
==> Dependencies
Build: autoconf ✔, automake ✘, libtool ✔
Optional: python ✘
==> Options
--with-python
	Build with python support
--with-test
	Run build-time check
--without-python@2
	Build without python2 support
--HEAD
	Install HEAD version
==> Caveats
Editor support and examples have been installed to:
  /usr/local/opt/protobuf/share/doc/protobuf
`

`$ brew search protobuf
==> Searching local taps...
protobuf ✔       protobuf@2.5 ✔   protobuf@2.6 ✔   protobuf@3.1 ✔   protobuf-c       protobuf-swift   swift-protobuf
==> Searching taps on GitHub...
homebrew/php/php53-protobuf             homebrew/php/php55-protobuf             homebrew/php/php71-protobuf
homebrew/php/php54-protobuf             homebrew/php/php56-protobuf
==> Searching blacklisted, migrated and deleted formulae...`

Maybe some check in the configure script would be handy.

kismetwireless, 2018-03-14:
It looks like you'll have to recompile the protobuf code to match your version; it looks like the OSX ports maintainers ship a much older version of protobufs; I've re-spinned the http protobuf to build on the old protobuf and protobuf-c on osx; for now you can do 'make protobuf-parsers'; i'll see how to detect it.

I'll leave this bug open for now.

mo8Zomo0, 2018-03-14:
thx for the quick response!

sadly I still struggle.
`make clean ; CFLAGS="-I/opt/local/include" LDFLAGS="-L/opt/local/lib" CPPFLAGS="-I/opt/local/include" ./configure`

ends with:

```
checking pkg-config is at least version 0.9.0... yes
checking for protobuf... yes
checking for libprotobufc... no
configure: error: missing libprotobuf-c

```
I am on:
```
$ date ;git status ; git log --pretty=format:'%h' -n 1
Wed Mar 14 20:25:32 CET 2018
On branch master
Your branch is up-to-date with 'origin/master'.

nothing to commit, working tree clean
d032c652
```

mo8Zomo0, 2018-03-14:
My fault, ...

```
$ brew install protobuf-c
Updating Homebrew...
```

lets see, ...

kismetwireless, 2018-03-14:
I just compiled a custom local version of protobuf 2.6.1 and am using it now to compile the protobuf parsers in git; you should be able to do a git pull of the absolute latest commit (as of a few minutes ago) and have it work; I can confirm at least that it compiles against the 2.6.1 protobuf in 'ports'; it should equally work with homebrew; you'll get some warnings from the protobuf lib itself but otherwise it builds clean.

I also updated the osx howto to have the new deps.

mo8Zomo0, 2018-03-14:
After me installing the protobuf-c the configure worked fine as well as the compile itself, thx!

closed 2018-03-14 by mo8Zomo0

kismetwireless, 2018-03-15:
For any who find this in the future; it looks like protobuf doesn't coexist nicely across versions at all; clearly the new compiled protobuf can't run with the old runtime, but the old runtime can't be built with the new environment, either.  I've shifted to requiring protoc and protoc-c to compile the protobuf components at build time.