kismetwireless/kismet #40: kismet dies after around 10 minutes

mo8Zomo0, 2018-03-14:
Hi!

I am on d032c652 and the process dies on me after around 10 minutes
(MacBook Air Mid 2011, with its integrated wifi running MacOS Sierra 10.12.6)

starting kismet with:
`sudo kismet -c en0 --debug 2>&1 |ts |tee -a kismet-m.log`

the last I see is:

```
Mar 14 21:23:49 INFO: Detected new 802.11 Wi-Fi device xx:xx:xx:xx:xx:xx packet 207556
Mar 14 21:23:51 FATAL:  Error during write(): Broken pipe
Mar 14 21:23:51 think we can set monitor
```

the two week old version also died, that was its output:
`ERROR: Attempting to log devices, but devices are still being saved from
       the last logging attempt.  It's possible your system is slow or you
       have a very large number of devices to log.  Try increasing the
       delay in 'kis_log_storage_rate' in kismet_logging.conf
ALERT: PACKETLOST Kismet has started to drop packets; the packet queue has
       a backlog of 8193 packets.  Your system may not be fast enough to
       process the number of packets being seen. You change this behavior
       in 'kismet_memory.conf'.
FATAL - Capture source did not get PING from Kismet for over 15 seconds; shutting down
think we can set monitor`

asmodianx, 2018-03-16:
Have you tried using the syslog plugin? Also are the default xml and text file logs insufficient for your needs?

mo8Zomo0, 2018-03-29:
Hi! The only machine involved is the mentioned mac book air where I locally run.
I did not try the syslog or any other plugins. I used more or less the default config.

kismetwireless, 2018-04-23:
This should have been resolved some weeks ago, either by the TCP code rewrite to use send/recv on tcp sockets, or in some threading fixing.

closed 2018-04-23 by kismetwireless