kismetwireless/kismet #41: kismet rtl433 capture - error message

asmodianx, 2018-03-16:
Traceback (most recent call last):
  File "/usr/local/bin/kismet_cap_sdr_rtl433", line 426, in <module>
    rtl.rtl_loop()
  File "/usr/local/bin/kismet_cap_sdr_rtl433", line 368, in rtl_loop
    if self.prep_kismet():
  File "/usr/local/bin/kismet_cap_sdr_rtl433", line 288, in prep_kismet
    d_created = rtl.create_datasource()
  File "/usr/local/bin/kismet_cap_sdr_rtl433", line 201, in create_datasource
    if self.debug:
AttributeError: kismet_rtl433 instance has no attribute 'debug'

ALSO: It would also help to have a list of python dependencies in the README.txt file in the capture directory.

asmodianx, 2018-03-22:
**I pulled it down today and its still giving me the error.**
_MQTT found; enabling MQTT features
Traceback (most recent call last):
  File "/usr/local/bin/kismet_cap_sdr_rtl433", line 426, in <module>
    rtl.rtl_loop()
  File "/usr/local/bin/kismet_cap_sdr_rtl433", line 368, in rtl_loop
    if self.prep_kismet():
  File "/usr/local/bin/kismet_cap_sdr_rtl433", line 288, in prep_kismet
    d_created = rtl.create_datasource()
  File "/usr/local/bin/kismet_cap_sdr_rtl433", line 201, in create_datasource
    if self.debug:
AttributeError: kismet_rtl433 instance has no attribute 'debug'_

So there is no object existing called "debug" (im not a python programer).  I removed the if statement on Line 201 and 202 and it works.

/usr/local/bin/kismet_cap_sdr_rtl433 --user admin --password ********* --name rtl433_capture --rtl433 /usr/local/bin/rtl_433 --device 0

kismetwireless, 2018-03-24:
I'll address this in the upcoming rewrite of the rtl433 code...

closed 2018-03-24 by kismetwireless