kismetwireless/kismet #43: when running kismet as a service error: "Error opening terminal: unknown."

asmodianx, 2018-03-22:
Is there a command-line option or "--quiet" to allow this to run as a service?
kismet_server has been discontinued/disabled which given the current direction of the project of making kismet a web service is kind of self defeating if it cant load at boot because it wants a user interactive console.

All I see are options to disable logging

> root@wifi-sensor:/etc/systemd/system# kismet --help
> Kismet server terminated.  Last output:
> 
>  *** Logging Options ***
>  -T, --log-types <types>      Override activated log types
>  -t, --log-title <title>      Override default log title
>  -p, --log-prefix <prefix>    Directory to store log files
>  -n, --no-logging             Disable logging entirely
> 
>  *** Device Tracking Options ***
>      --device-timeout=n       Expire devices after N seconds
> Kismet exited.
> root@wifi-sensor:/etc/systemd/system#
>

kismetwireless, 2018-03-24:
--no-ncurses is what you're looking for most likely

asmodianx, 2018-03-29:
Great that worked!
can some one add this to the documentation and the "--help" command?

kismetwireless, 2018-07-27:
Added to readme

closed 2018-07-27 by kismetwireless