kismetwireless/kismet #51: State frames (auth/assoc) included in the station pcap

lordberre, 2018-04-19:
Hi,
First of all I have to say fantastic work with kismet.

The handshake capture is good, however, it would be very useful to capture the subsequent auth/(re)assoc frames as well, as it is very useful when debugging roaming and such. Can anyone familiar with the code estimate how much work this would be?

Kind Regards,
B

kismetwireless, 2018-04-19:
This would be better handled by timeslicing the kismet log or the pcap log files; keeping too many packets in ram will quickly bloat kismet per-network records; this could be sovled via the future kismet-log REST api, which is not yet complete.

lordberre, 2018-04-22:
You're right about the fact that I can cut this from the pcaps/logs. I can probably do something nice with a some filtering in tshark or similar. I just thought would be generally useful to have along with the 4way handshake as it could be very useful at times.

Thanks

closed 2018-04-22 by lordberre