kismetwireless/kismet #62: Errors when building Kismet for OpenWRT

adde88, 2018-07-06:
Hi there, and thanks for building this awesome tool!

I've been working on keeping Kismet up-to-date with your latest commits for OpenWRT.
It's been a while since i last updated it, so i sat down today to update both kismet, and kismet-remote.

**kismet-remote** builds without perfectly, without any issues on the latest commit (6.June 2018)
However, building **kismet** fails with the following error on the latest commit:
```
trackedelement.h: In member function 'virtual void TrackerElementCoreNumeric<N>::coercive_set(const string&)':
trackedelement.h:549:18: error: 'stod' is not a member of 'std'
         auto d = std::stod(in_str);
                  ^
Makefile:476: recipe for target 'util.cc.o' failed
make[3]: *** [util.cc.o] Error 1
```

I then tested building older commits, and it seems like issues started to happen on the first commit on June 22. (195a647bc66de6362900186cd24784b762f74a4b)
As the earlier commit from June 21(4b71d0daab317b9fac2ef219dcfffa22bb4022b4) builds fine.
Every other commit after that fails.

I used git bisect on the one that works, and the one that fails, and this is the output from bisect:
```
195a647bc66de6362900186cd24784b762f74a4b is the first bad commit
commit 195a647bc66de6362900186cd24784b762f74a4b
Author: Mike Kershaw / Dragorn <dragorn@kismetwireless.net>
Date:   Thu Jun 21 23:42:21 2018 -0400

    Start major rewrite of internal structures to eliminate shared union and
    make the whole system simpler to use and more C++11-idiomed

:100644 100644 7b2b8e9d8088a46505b76088f2d8083cd94de4af de799a03bcf7d2ccccc0803fe183f7d52fed4e68 M	trackedelement.cc
:100644 100644 beed7ab0a21401263e41e704c367a92b28dd2427 0da558d8f331b20241acce1938f36f60d4d820a4 M	trackedelement.h
```

adde88, 2018-07-07:
I forgot to mention that there also was some warnings produced, right before the error.
I'm not sure if they're related in any way, so i thought i'd post the output with the warnings and the error if it helps in any way with deugging:
```
ccache_cxx -std=gnu++11 -Wall -g -I. -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float  -pthread -I/root/openwrt-sdk-testing/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include  -I/root/openwrt-sdk-testing/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/root/openwrt-sdk-testing/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/root/openwrt-sdk-testing/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/root/openwrt-sdk-testing/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include   -c util.cc -o util.cc.o

In file included from packet.h:38:0,
                 from util.cc:66:
packet_ieee80211.h:270:40: warning: 'packed' attribute ignored [-Wattributes]
 } frame_control __attribute__ ((packed));
                                        ^
packet_ieee80211.h:275:43: warning: 'packed' attribute ignored [-Wattributes]
 } wireless_fragseq __attribute__ ((packed));
                                           ^
packet_ieee80211.h:295:43: warning: 'packed' attribute ignored [-Wattributes]
 } fixed_parameters __attribute__ ((packed));
                                           ^
In file included from packet.h:39:0,
                 from util.cc:66:
trackedelement.h: In member function 'virtual void TrackerElementCoreNumeric<N>::coercive_set(const string&)':
trackedelement.h:557:18: error: 'stod' is not a member of 'std'
         auto d = std::stod(in_str);
                  ^
Makefile:476: recipe for target 'util.cc.o' failed
make[3]: *** [util.cc.o] Error 1
```

adde88, 2018-07-07:
I did some research and the error: ```error: 'stod' is not a member of 'std'``` seems to usually happen when people don't declare -std=c++11 or -std=gnu++11.

But your project obviously does this!
So, the error might come from my compiler, or libraries like libstc++ and/or libc++ (Not sure yet, which one of them my SDK uses.)
I'll do some more testing. And let you know if i find anything to report back to you on. :)

kismetwireless, 2018-07-15:
Some compilers are just broken; the older ones in some openwrt flavors, debian, and ubuntu 14 are the main culprits.

I just removed stod entirely and used a stringstream to do a conversion to double; that function is rarely called so the inefficiency shouldn't be particularly terrible.

Latest commits OUGHT to compile on openwrt now (or at least, fail somewhere else now); i haven't gotten a chance to test myself yet.

adde88, 2018-07-16:
Thanks for updating me. I'll test a compilation asap. :) :+1:

adde88, 2018-07-18:
Yeah, the issue seems to be fixed by those changes. It's compiling just fine now. Thanks alot. :)

closed 2018-07-18 by adde88