kismetwireless/kismet #64: Time Stamp in nanoseconds

leonidas183, 2018-07-10:
Hi,

Is it possible to set the time stamp precision in nanoseconds, something similar to tcpdump -time-stamp-precision nano ? 

I saw there is a commit that is saying "Add optional high-precision time field for GPS", is it related to what I want ? If yes, how do I enable this option? 

I also tried to search how the timestamp is set if it could help to answer my question and i think it is with Timetracker (https://github.com/kismetwireless/kismet/blob/master/timetracker.h) which uses timeval (and timeval is in microseconds according to the definition).

kismetwireless, 2018-07-11:
I'll put it on the list, but it's unlikely to be high on the list - none of the wifi cards support ns timestamping so there isn't really much benefit to it in terms of TDIF analysis, etc.

leonidas183, 2018-07-11:
What you mean by "none of the wifi cards support ns timestamping" ? 

I thought that the timestamp set in pcapdump files was from the OS and what I need is to have it with nanoseconds precision.

kismetwireless, 2018-07-11:
Timestamp comes from the pcap stream; nanosecond is an optional variant of pcap which is only supported on limited physical interfaces.  No wi-fi driver I'm aware of supports nanosecond precision capture.

By the time you get to the kernel layer, you're at microseconds precision already because you have no control over how long it takes to get from radio->decoder->dma/mailbox->interface (pci or usb)->interface interrupt->kernel skb.

If there re any wifi drivers with phy nanosecond timestamping I'll consider pushing this higher up the list, in the meantime patches are welcome but they'll have to be wide-ranging - all pcap creation and reading code, all timestamp fields in all areas, a new flag in the RPC protocol indicating MS is now NS, and probably some other spots too.