kismetwireless/kismet #67: log-type PCAPNG  Crashes Kismet

bioless, 2018-07-12:
[gdb.txt](https://github.com/kismetwireless/kismet/files/2189373/gdb.txt)

Ran the command below.  Additionally, I have even tried different wireless adapters. It fails if I try changing the configuration and/or add it in the command line.   pcappppi works fine.

I'm running it on the latest Kali updated. I followed the instructions, but I did have to add the following drivers for one of my wireless cards. realtek-rtl88xxau-dkms.  Additionally I've tested capturing over wireshark just fine.

➜  ~ kismet -c wlan0mon --log-types pcapng 
Kismet server terminated.  Last output:
/opt/kismet/pcapng_stream_ringbuf.cc", line 334, in pcapng_write_packet
               414: #ifndef DISABLE_BACKWARD
        415:     using namespace backward;
      > 416:     StackTrace st; st.load_here(32);
        417:     Printer p; p.print(st);
        418: #endif
Kismet exited.

Thanks for the great work!

kismetwireless, 2018-07-12:
Looks like it's a threading issue caused in the latest multithreaded packet decode code, I'll look into it.

bioless, 2018-07-12:
No more crashing, PCAP opens with Wireshark. Thank you.

closed 2018-07-12 by bioless