kismetwireless/kismet #72: Encryption Type Column

ryanvillarreal, 2018-07-31:
Sorry if this is a repeat or bad question, but I have been digging through the code trying to figure out where to add an Encryption Type column to the main dashboard.  Any help would be appreciated.

kismetwireless, 2018-08-05:
You could display the device-level crypttype string but you'd probably have to update all the phys to update the cryptstring field; right now nothing populates that but it's on the list.

ryanvillarreal, 2018-08-07:
Alright thanks for the update.  I'll close then.

closed 2018-08-07 by ryanvillarreal

kismetwireless, 2018-08-19:
I just added a simplified wifi crypt field and populated the generic crypt field with it, and added a column - if that's what you needed, there you go, and if you wante dto expand it from there, that's a much more solid basis to start from.

Hope that helps!

ryanvillarreal, 2018-08-20:
Just pulled the latest and recompiled source.  This is beautiful.  Thank you so much!