kismetwireless/kismet #75: Crashing when trying to save handshake from browser

kallakoras, 2018-08-20:
Kismet 2018-08-BETA1

Latest git pull, should be up to date.

Captured 4 way handshake and try to download it via web browser, immediately crashes the server with the attached output.

What am I doing wrong here?

``Kismet server terminated.  Last output:
]
        513:         for (unsigned int i = 0; i < kishttpd->handler_vec.size(); i++) {
        514:             Kis_Net_Httpd_Handler *h = kishttpd->handler_vec[i];
        515:
      > 516:             if (h->Httpd_VerifyPath(url, method)) {
        517:                 handler = h;
        518:                 break;
        519:             }
#2    Source "/root/.scripts/kismet/phy_80211.cc", line 2401, in Httpd_VerifyPath [0x56301e72a0e5]
       2398:         return "WEP104";
       2399:
       2400:    if (cryptset & crypt_tkip)
      >2401:         return "WPA-TKIP";
       2402:
       2403:    if (cryptset & crypt_aes_ocb)
       2404:         return "WPA-OCB";
#1    Source "/root/.scripts/kismet/devicetracker.cc", line 502, in FetchDevice [0x56301e6d6c9f]
        499: }
        500:
        501: std::shared_ptr<kis_tracked_device_base> Devicetracker::FetchDevice(device_key in_key) {
      > 502:     local_locker lock(&devicelist_mutex);
        503:
        504:    device_itr i = tracked_map.find(in_key);
#0  | Source "/root/.scripts/kismet/kis_mutex.h", line 212, in try_lock_for
    |   210:         cpplock.lock();
    |   211: #else
    | > 212:         if (!cpplock.try_lock_for(std::chrono::seconds(KIS_THREAD_DEADLOCK_TIMEOUT))) {
    |   213:             throw(std::runtime_error(fmt::format("deadlock: mutex not available within "
    |   214:                             "{}", KIS_THREAD_DEADLOCK_TIMEOUT)));
      Source "/root/.scripts/kismet/kis_mutex.h", line 108, in local_locker [0x56301e5ae08a]
        105:         owner_count(-1) { }
        106:
        107:     bool try_lock_for(const std::chrono::seconds& d) {
      > 108:         if (std::this_thread::get_id() == owner) {
        109:             owner_count++;
        110:         } else {
        111:             mutex.try_lock_for(d);
Unknown signal -1616110384
watf? exit
Kismet exited.``

a9e78rauijasdf, 2018-08-30:
Can also reproduce the problem.

kismetwireless, 2018-10-03:
Should be addressed in the latest commit

closed 2018-10-03 by kismetwireless