kismetwireless/kismet #77: Feature request - Select /Deselect All channels

birukun, 2018-08-25:
In the Capture Sources menu, is it possible to make the 'All' button a toggle for channels?  ie, Set all channels / clear all channels when using the button.

I often use many adapters and provision certain channels to adapters, and the dual band cards I have require me to deselect a lot of channels.  It would be awesome if the button would clear all the channels allowing for quick selection of a few.

ThingEngineer, 2019-09-27:
Or a simple band selection such as that implemented in aircrack-ng i.e.
airodump-ng wlan1mon --band a

Thus only selecting 5GHz channels.

BeingDatta, 2020-04-08:
Yes, I also think it should be done in such a manner that we can select the channels manually. At present the options are to lock kismet on only one channel or enable on all channels which is not a problem with 2.4ghz adapter because it has only 14 channels but for 5ghz adapter there are so many channels and its nearly irritating and frustrating to deselect them individually if we don't need them. There should be a way to lock multiple channels manually.

Please implement this feature asap.

Regards.