kismetwireless/kismet #78: Errors running Kismet on Raspberry Pi 3

bluepuma77, 2018-08-30:
I compiled and installed Kismet according to documentation.

When trying to run Kismet on the latest Raspbian on a Raspi 3 I get errors:

```
pi@pi3:~/kismet $ sudo ./kismet -c wlan0
Kismet server terminated.  Last output:
ERROR: No helper_binary_path found in kismet.conf, make sure your config
       files are up to date; using the default binary path where Kismet is
       installed.
ERROR: No helper_binary_path found in kismet.conf, make sure your config
       files are up to date; using the default binary path where Kismet is
       installed.
ERROR: No helper_binary_path found in kismet.conf, make sure your config
       files are up to date; using the default binary path where Kismet is
       installed.
#1    Source "../sysdeps/unix/sysv/linux/arm/sigrestorer.S", line 64, in $a [0x766eb6af]
#0  | Source "/home/pi/kismet/backward.h", line 2046, in load_from
    |  2044: #endif
    |  2045: 		if (error_addr) {
    | >2046: 			st.load_from(error_addr, 32);
    |  2047: 		} else {
    |  2048: 			st.load_here(32);
      Source "/home/pi/kismet/backward.h", line 631, in sig_handler [0x34ec67]
        628: 		return size();
        629: 	}
        630: 	size_t load_from(void* addr, size_t depth=32) {
      > 631: 		load_here(depth + 8);
        632:
        633: 		for (size_t i = 0; i < _stacktrace.size(); ++i) {
        634: 			if (_stacktrace[i] == addr) {
Kismet exited.
```
Sometimes
```
pi@pi3:~/kismet $ sudo ./kismet -c wlan0
Kismet server terminated.  Last output:
ERROR: Missing 'ouifile' option in config, will not resolve manufacturer
       names for MAC addresses
INFO: Kismet has generated a random login for the web UI; it has been
      saved in /kismet_httpd.conf.
ALERT: NEWHTTPDUSER LK|vLK|vX�CX�Ct time you have run Kismet on this
       account.  A new password has been automatically generated, and is
       in /kismet_httpd.conf.  You will need this password to configure
       Kismet from the web interface.
Uncaught exception "std::bad_alloc"
Stack trace (most recent call last):
#1    Object "/usr/lib/arm-linux-gnueabihf/libstdc++.so.6.0.22, at 0x7691fdaf, in
#0    Source "/home/pi/kismet/kismet_server.cc", line 414, in TerminationHandler
        412: #ifndef DISABLE_BACKWARD
        413:     using namespace backward;
      > 414:     StackTrace st; st.load_here(32);
        415:     Printer p; p.print(st);
        416: #endif
Kismet exited.
```
Any idea what went wrong?

kbeflo, 2018-09-22:
Kismet fails to run because you're running it from inside the `~/kismet directory`, Kismet looks for the configuration files on `/usr/local/etc/` and won't find them, probably because you haven't installed them properly yet. 

Check out the readme section again and make sure to have followed everything and make sure you have really finished install, you could have just missed some commands.

ChrisLundquist, 2018-10-04:
I didn't think raspbian supported monitor mode for brcmfmac and you had to run kali for the firmware fixes and driver patches?

https://github.com/torvalds/linux/commit/01f69dfafdbe7deff58b58053bc3a4a75c6a570c#diff-1b8fac2d29272755e8f53049866fc1e5
https://github.com/torvalds/linux/commit/e63410ac65e0ead2040bbd3927c116889edf87e4#diff-1b8fac2d29272755e8f53049866fc1e5
https://github.com/torvalds/linux/commit/a8d7631858aff156b72f807ee7cc062048e63836#diff-1b8fac2d29272755e8f53049866fc1e5

I think you need these commits along with some other jazz, which are in mainline 4.18, but haven't been ported to https://github.com/raspberrypi/linux/ just yet.

If I'm wrong and all this should "just work" that's even more awesome!

ChrisLundquist, 2018-10-04:
> @ChrisLundquist <https://github.com/ChrisLundquist> Kali Linux Raspberry
> Pi arm images with nexmon has these driver patches included.

Right, that’s what I was trying to say :)
 @bluepuma77 said he was on raspbian latest though.

closed 2019-07-02 by kismetwireless