kismetwireless/kismet #90: Error during make (OS X Mojave)

Argetan, 2018-10-24:
After doing `./configure` (no errors as far as I can tell) I ran `make` which threw the following error:
```
Undefined symbols for architecture x86_64:
  "void boost_like::hash_combine<unsigned long>(XXHash32&, unsigned long const&)", referenced from:
      Kis_80211_Phy::HandleSSID(std::__1::shared_ptr<kis_tracked_device_base>, std::__1::shared_ptr<dot11_tracked_device>, kis_packet*, dot11_packinfo*, kis_gps_packinfo*) in phy_80211.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [kismet] Error 1
```

I'm guessing a framework or library is missing, however, I'm unable to tell which one. My version of `clang` is the following in case it matters:
```
Apple LLVM version 10.0.0 (clang-1000.11.45.2)
Target: x86_64-apple-darwin18.0.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

kismetwireless, 2018-10-24:
Make sure you've got the latest git pull and do a `make distclean` and
reconfigure; I'd guess you've got a stale makefile; that code changed
recently and moved to another file, so if you have a stale makefile it
won't get linked and you'll get that error.

Argetan, 2018-10-24:
Created a fresh directory and cloned the `master` branch, ran `./configure` again but the error remains.

kismetwireless, 2018-10-24:
Unfortunately I don't have any other ideas right now, and I don't have
access to any mojave systems; I've done a clean checkout on several other
systems and that file is linking properly; I'll keep digging at it though.

kismetwireless, 2018-10-25:
> I just tried compiling on a Mac running Sierra and it threw the same
> error, could it be that both systems are missing a library/framework
> somewhere?

Try the latest commit

Argetan, 2018-10-25:
It compiled successfully on both systems, thank you for the help!

closed 2018-10-25 by Argetan

thewifimaster, 2018-10-25:
I get an error 'No package 'protobuf‘ found‘.

How did you install https://github.com/protocolbuffers/protobuf/releases to get it to work?

Argetan, 2018-10-25:
I think I already had it installed but you should be able to install it using HomeBrew:
`brew install protobuf`

thewifimaster, 2018-10-25:
That is what I did and I still get the error. Can’t seem to figure it out as it must be in standard paths .. Any thoughts?

Argetan, 2018-10-25:
`brew link protobuf` perhaps? Might have to unlink it first but Brew will tell you in that case.

thewifimaster, 2018-10-25:
No success with that. Any other ideas?

DaniloNC, 2018-12-27:
@thewifimaster 
Try: ```brew install protobuf-c```