kismetwireless/kismet #92: Feature request - MAC filtering

dibg, 2018-10-30:
Android phone and ios devices in their attempt to anonymize the user while is not connected to a network broadcast with a random mac address which changes often. As a result we end up having thousand  random devices in the capture data. Slowing down the gui and the ability to effectively search though the data. Essentially is just noise.  Filtering this devices before capturing is a necessity.

A example of random mac list:
DA:A1:19:00:01:22
DA:A1:19:31:4D:12
DA:A1:19:A0:FE:62

Therefore implementing a capture filter option with regular expression capabilities should be able to eliminated them.

More info  about the randomization:
https://source.android.com/devices/tech/connect/wifi-mac-randomization

stipe42, 2019-05-02:
How can it be solved with regular expression only, it is far more complex.
http://papers.mathyvanhoef.com/wisec2016.pdf

kismetwireless, 2019-05-03:
Filtering support is present in 2019-04; regex of mac addresses is not, and
is unlikely to ever be feasible.  Filtering has to be lightning fast when
applied to thousands of packets a second; regex cannot keep up with that.

Filtering supports matching by OUI (or other groups of addresses) however.

https://www.kismetwireless.net/docs/readme/logging/#filtering

hfreire, 2019-08-05:
@kismetwireless in the documentation you pointed out, it states:

> Filtering device records from the kismetdb log will not prevent Kismet from showing them, but prevents Kismet from logging them.

Will this still display in the webpage these random MAC addresses that @dibg pointed out? Is there any other way of filtering out a block of MAC addresses from ever being displayed in the webpage?

kismetwireless, 2019-08-05:
Currently, no; I'm still formulating how display filtering will work.

elixx, 2019-10-30:
Not sure if this should be a separate issue, but the old gen code allowed for completely filtering out a SSID by name.  I'd really like that again, it was fairly necessary a lot of times.

emory, 2021-06-24:
hi, people! first of all thank you to everyone that works on this project, i really have enjoyed watching kismet evolve over the years. i think i used it for my first WiFi pentest engagements back in the early 2000s but it's definitely come a long way since then! thank you 🙇🏻 

obviously apple has embraced MAC randomization and I tend to accumulate like 16,000 MAC addresses over the course of a weekend at either residence and i've been considering options on how to treat them. i know the ranges that are used at least by Apple for their implementations, but i don't know if it is best to collect those addresses and corral them into a category i can easily hide/ignore, or if not recording them in the first place is the best way to do it, but i think there is still useful information that can be gained from random or psuedo-random devices beaconing and probing.

i'm going to supplement my passive WiFi collection with Bluetooth data collection as well and then i'll have to re-assess what I'd like to do with this information. i can haul the data out of the kismet database, transform it into a CSV file i can slice and dice and chop out to exclude ranges of addresses i know aren't global unique, but then i've still got low-spec systems handling thousands of records every hour or so and it's not immediately useful data to me (yet?).

i am hoping to get some direction on where my attention should be focused; doing a collection filter or plugin to filter/sift and segregate random/psuedo-random MACs, ignore them entirely, or strip them out as noise when reporting? i think i can get some context but i don't have sufficient data yet and i still have to think about this for myself but would appreciate some feedback on where Kismet is headed with this sort of thing before i start making changes.

where i am right now i've got about 35 devices in my house and half of them are macOS/iPad/iOS/tvOS, but obviously it's those neighbors and my own devices beaconing occasionally that is burying me. i don't know if i should open a separate issue for this but this one seemed pretty close to what i have been thinking about. //@emory