kismetwireless/kismet #94: Submitting .kismet files to Wigle.net

kbeflo, 2018-11-17:
Using a G-STAR IV GPS and Ralink RT5370's on a Raspberry Pi 3 running Raspbian with gpsd, I tried submitting a couple of .kismet files to Wigle.net. The files are processed but no New WiFi and GPS locations found. 

On the Kismet web interface GPS is green and GPS fixed with coordinates while running.

kismetwireless, 2018-11-17:
I don't believe wigle has processing for the kismetdb format currently; now that the file format is getting nailed down, I'll talk to them about the possibility of adding it on their end.

In the meantime you could try to convert it to kml or some other format wigle accepts.

closed 2018-11-17 by kismetwireless

kbeflo, 2018-11-26:
@kismetwireless I've tried every log format, none got accepted. Going to use Kismet 2016-07-R1 for now.

designer2k2, 2019-07-11:
use the kismetdb_to_wiglecsv tool installed with latest kismet to convert the .kismet file to csv that works with wigle.