kismetwireless/kismet #97: [OpenWRT CC - ar71xx] - Errors when compiling latest commit

adde88, 2018-12-01:
Trying to cross-compile the latest commit using OpenWRT 15.05 SDK. (ar71xx) 
Gives this output right after ./configure has been completed.
```
Configuration complete:
         Compiling for: linux-gnu (mips)
           C++ Library: stdc++
   Installing as group: root
    Man pages owned by:
       Installing into: /usr
          Setuid group: kismet
   Linux WEXT capture : yes
        Prelude  SIEM : no
   PCRE Regex Filters : no
          pcap capture: yes
       airpcap control: n/a (only Cygwin/Win32)
        PPI log format: yes
LibCapability (enhanced
   privilege dropping): no
   OSX/Darwin capture : n/a (only OSX/Darwin)
         Linux Netlink: yes (mac80211 VAP creation) - libnl-3.0 libnl-genl-3.0
  Linux NetworkManager: no (will not be able to control NetworkManager)
       Linux Bluetooth: yes
         nRF MOuseJack: yes
        Python Modules: no (will not build Python modules and related support)
       RTL-SDR RTL_433: no (python modules are disabled)
      Freaklabs Zigbee: no (python modules are disabled)
 lm-sensors monitoring: no (will not be able to monitor system temperature, etc)
        Built-in Debug: yes - Full debug info available on crash, using unwind

*** WARNING ***
libnm (the NetworkManager control library) was not found.  Kismet uses
libnm to prevent NetworkManager from reconfiguring monitor mode interfaces
while Kismet is running; without this library you will need to manually
disable NetworkManager or manually tell it to ignore specific interaces.
rm -f /root/openwrt-sdk-2/build_dir/target-mips_34kc_uClibc-0.9.33.2/kismet-01.12.2018/.configured_*
touch /root/openwrt-sdk-2/build_dir/target-mips_34kc_uClibc-0.9.33.2/kismet-01.12.2018/.configured_yynn
make -C /root/openwrt-sdk-2/build_dir/target-mips_34kc_uClibc-0.9.33.2/kismet-01.12.2018 LD="ccache_cxx" all
make[3]: Entering directory '/root/openwrt-sdk-2/build_dir/target-mips_34kc_uClibc-0.9.33.2/kismet-01.12.2018'
/root/openwrt-sdk-2/staging_dir/host/bin/protoc --cpp_out=./protobuf_cpp/ -I protobuf_definitions/ protobuf_definitions/kismet.proto
/root/openwrt-sdk-2/staging_dir/host/bin/protoc --cpp_out=./protobuf_cpp/ -I protobuf_definitions/ protobuf_definitions/http.proto
/root/openwrt-sdk-2/staging_dir/host/bin/protoc --cpp_out=./protobuf_cpp/ -I protobuf_definitions/ protobuf_definitions/datasource.proto
/root/openwrt-sdk-2/staging_dir/host/bin/protoc --cpp_out=./protobuf_cpp/ -I protobuf_definitions/ protobuf_definitions/linuxbluetooth.proto
/root/openwrt-sdk-2/staging_dir/host/bin/protoc --cpp_out=./protobuf_cpp/ -I protobuf_definitions/ protobuf_definitions/sdrrtl433.proto
/root/openwrt-sdk-2/staging_dir/host/bin/protoc --cpp_out=./protobuf_cpp/ -I protobuf_definitions/ protobuf_definitions/kismet.proto
ccache_cxx -std=gnu++11 -MM -Wall -g -I. -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16  -O3 -pthread -I/root/openwrt-sdk-2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include  -I/root/openwrt-sdk-2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/root/openwrt-sdk-2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/root/openwrt-sdk-2/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/root/openwrt-sdk-2/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include   protobuf_cpp/kismet.pb.cc | sed -e "s/\.o/\.cc.o/" > protobuf_cpp/kismet.pb.cc.d
ccache_cxx -std=gnu++11 -Wall -g -I. -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16  -O3 -pthread -I/root/openwrt-sdk-2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include  -I/root/openwrt-sdk-2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/root/openwrt-sdk-2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/root/openwrt-sdk-2/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/root/openwrt-sdk-2/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include   -c protobuf_cpp/kismet.pb.cc -o protobuf_cpp/kismet.pb.cc.o
{standard input}: Assembler messages:
{standard input}:5792: Error: unrecognized opcode `sync'
{standard input}:5999: Error: unrecognized opcode `sync'
{standard input}:6204: Error: unrecognized opcode `sync'
{standard input}:6409: Error: unrecognized opcode `sync'
{standard input}:6614: Error: unrecognized opcode `sync'
{standard input}:6819: Error: unrecognized opcode `sync'
{standard input}:10027: Error: unrecognized opcode `sync'
{standard input}:10335: Error: unrecognized opcode `sync'
{standard input}:10643: Error: unrecognized opcode `sync'
{standard input}:10955: Error: unrecognized opcode `sync'
{standard input}:15409: Error: unrecognized opcode `sync'
{standard input}:15718: Error: unrecognized opcode `sync'
{standard input}:16416: Error: unrecognized opcode `sync'
{standard input}:16974: Error: unrecognized opcode `sync'
{standard input}:18073: Error: unrecognized opcode `sync'
{standard input}:18371: Error: unrecognized opcode `sync'
{standard input}:18680: Error: unrecognized opcode `sync'
{standard input}:19270: Error: unrecognized opcode `sync'
{standard input}:19661: Error: unrecognized opcode `sync'
{standard input}:21009: Error: unrecognized opcode `sync'
{standard input}:21968: Error: unrecognized opcode `sync'
{standard input}:22600: Error: unrecognized opcode `sync'
{standard input}:22905: Error: unrecognized opcode `sync'
{standard input}:23428: Error: unrecognized opcode `sync'
{standard input}:23815: Error: unrecognized opcode `sync'
{standard input}:24646: Error: unrecognized opcode `sync'
{standard input}:25727: Error: unrecognized opcode `sync'
{standard input}:27161: Error: unrecognized opcode `sync'
{standard input}:27684: Error: unrecognized opcode `sync'
{standard input}:28071: Error: unrecognized opcode `sync'
make[3]: *** [Makefile:505: protobuf_cpp/kismet.pb.cc.o] Error 1
make[3]: Leaving directory '/root/openwrt-sdk-2/build_dir/target-mips_34kc_uClibc-0.9.33.2/kismet-01.12.2018'
make[2]: *** [Makefile:85: /root/openwrt-sdk-2/build_dir/target-mips_34kc_uClibc-0.9.33.2/kismet-01.12.2018/.built] Error 2
make[2]: Leaving directory '/root/gits/openwrt-useful-tools/kismet-Makefiles-openwrt-cc/package/net/kismet'
make[1]: *** [package/Makefile:192: package/net/kismet/compile] Error 2
make[1]: Leaving directory '/root/openwrt-sdk-2'
```

I've attempted to play with the Makefile ALOT!
And **PKG_USE_MIPS16:=0** is already being used, which is normally the cause of these error messages.

Any ideas on how i can solve this?

nzamps, 2019-01-08:
Not that helpful but I just compiled BETA 1 on ar71xx using the Makefile from https://github.com/kismetwireless/kismet/blob/master/packaging/openwrt/kismet-2018/Makefile

Went ok - only change I had to add was **--disable-libusb** in the CONFIGURE_ARGS

One thing I noticed, comparing your compile, is that I don't see the use of ccache_cxx but mips-openwrt-linux-musl-g++

adde88, 2019-01-11:
The solution to my error message was adding PKG_USE_MIPS16:=0 to the Makefile.
However, this gave me a new error after a while:
```
ccache_cxx -std=gnu++11 -Wall -g -I. -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float  -O3 -pthread -I/root/openwrt-sdk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include  -I/root/openwrt-sdk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/root/openwrt-sdk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/root/openwrt-sdk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/root/openwrt-sdk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include   -c util.cc -o util.cc.o
In file included from globalregistry.h:29:0,
                 from packet.h:36,
                 from util.cc:71:
kis_mutex.h: In constructor 'local_shared_unlocker::local_shared_unlocker(kis_recursive_timed_mutex*)':
kis_mutex.h:372:20: error: invalid initialization of non-const reference of type 'kis_recursive_timed_mutex&' from an rvalue of type '<brace-enclosed initializer list>'
         cpplock{*in} { }
                    ^
kis_mutex.h: In constructor 'local_shared_unlocker::local_shared_unlocker(kis_recursive_timed_mutex&)':
kis_mutex.h:375:19: error: invalid initialization of non-const reference of type 'kis_recursive_timed_mutex&' from an rvalue of type '<brace-enclosed initializer list>'
         cpplock{in} { }
                   ^
make[3]: *** [Makefile:529: util.cc.o] Error 1
```
ccache_cxx is just another name for "mips-openwrt-linux-uclibc-g++" as the SDK has ccache enabled.

nzamps, 2019-01-11:
Ah yes, sorry, I was trying with 18.06.

adde88, 2019-01-29:
Closing this. Will send a PR with the Makefile + patches needed to compile with CC SDK.

closed 2019-01-29 by adde88