kismetwireless/kismet #100: Fails on Make - Kali OS raspberry pi

austino5308, 2018-12-24:
I am installing kismet on a raspberry pi 3 running kali linux.

./configure shows that everything is okay

`Configuration complete: 
         Compiling for: linux-gnu (aarch64)
           C++ Library: stdc++
   Installing as group: root
       Installing into: /usr/local
          Setuid group: kismet
   Linux WEXT capture : yes
        Prelude  SIEM : no
   PCRE Regex Filters : yes
          pcap capture: yes
       airpcap control: n/a (only Cygwin/Win32)
        PPI log format: yes
LibCapability (enhanced
   privilege dropping): yes
   OSX/Darwin capture : n/a (only OSX/Darwin)
         Linux Netlink: yes (mac80211 VAP creation) - libnl-3.0 libnl-genl-3.0
  Linux NetworkManager: yes (libnm NetworkManager control)
       Linux Bluetooth: yes
         nRF MOuseJack: yes
        Python Modules: yes
       RTL-SDR RTL_433: yes
        RTL-SDR RTLAMR: yes
       RTL-SDR RTLADSB: yes
      Freaklabs Zigbee: yes
 lm-sensors monitoring: yes
        Built-in Debug: yes - Full debug info available on crash, using unwind
`

but I get this error message on "make"
`root@kali:~/kismet# make
g++ -std=gnu++17 -Wall -g -I. -g -O2 -O3 -pthread -fopenmp  -c pipeclient.cc -o pipeclient.cc.o
pipeclient.cc:1:1: error: unterminated comment
 /*
 ^
make: *** [Makefile:535: pipeclient.cc.o] Error 1
`

Any help would be much appreciated!

kismetwireless, 2018-12-24:
You've most likely run out of RAM - running kismet doesn't take much, but compiling it takes a LOT.

I recommend NOT using -j in Make and letting it compile a single file at a time, and increasing your swap size significantly.

Your other option is compiling it out-of-body on a PC using something like systemd-nspawn and qemu-arm.

Packages for Kali on rpi will be coming in the near future, but are not ready yet.

austino5308, 2018-12-24:
I wasn't using -j in make. It failed the moment I ran make. I will have to take a look at attempting to compile it out-of-body. It's weird because I had kistmet compiled and working about a year ago and I abandoned the project I was working on at the time. When returning and starting fresh, I run into this. I just didn't know if you had any suggestions on what I could be doing wrong.

austino5308, 2018-12-24:
[image]

greaterthanstar, 2019-02-02:
I was having the same problem on a Raspberry Pi 3 with Kali. I would run `make` and it would fail with that "Error 1". If I ran `make` again it would fail again, but not always on the same line. Eventually I could not get past phys_80211.c and came here. I tried to follow dragorn's advice and increase the swap, but Kali apparently doesn't have that service installed by default! So here's the instructions to do it. I am assuming you are root when running these commands. 

Install the swap service
`apt install -y dphys-swapfile`

turn off the service so you can edit the config. 
`swapoff -a`

Open up the config file and increase the size.
`nano /etc/dphys-swapfile`

in that file, change the following line and save. Most things I read said to set it to 1024 but since dragorn said "significantly", I went big with it. 
`CONF_SWAPSIZE=2048`

Turn it back on.
`swapon`

Now try `make` again.

strasharo, 2019-02-19:
You don't need a whole service to do it.
Just cd into a place you have enough space to write (I usually do that on a mounted external USB flash drive) and create a swap file:
```
root@zero2:~# mount /dev/sda1 /mnt/flash
root@zero2:~# cd /mnt/flash/
root@zero2:/mnt/flash# fallocate -l 1G swapfile
root@zero2:/mnt/flash# 
```
Then format the file as swap:
```
root@zero2:/mnt/flash# mkswap swapfile 
mkswap: swapfile: insecure permissions 0644, 0600 suggested.
Setting up swapspace version 1, size = 1024 MiB (1073737728 bytes)
no label, UUID=ca52d84b-c546-4114-807e-ce4903a9d357
root@zero2:/mnt/flash# 

```
Apply the recommended permissions:
```
root@zero2:/mnt/flash# chmod 0600 swapfile 
root@zero2:/mnt/flash# 
```
Enable the swapfile:
```
root@zero2:/mnt/flash# swapon swapfile
root@zero2:/mnt/flash# 
```
Confirm that the new swap memory is available:
```
root@zero2:/mnt/flash# free -m
              total        used        free      shared  buff/cache   available
Mem:            433          27          95           5         310         389
Swap:          1023           0        1023
root@zero2:/mnt/flash# 
```
And that's it. Just swapon the extra swap memory when you need it. The change is not permanent, so on the next reboot it's going to be detached.

DiscoveringWorld, 2020-04-06:
Thanks i will try that on my rb0

closed 2022-06-17 by rc-austinoneil