kismetwireless/kismet #106: Tag a version for a Mac brew release

wrobelda, 2019-01-02:
Hi,

I am about to send a pull request for Homebrew to include the kismet formula in the official branch. They require an actual release (as opposed to git HEAD) to be used for source code, and the most recent kismet-2018-08-beta1 release does not build under macOS (https://github.com/kismetwireless/kismet/issues/80).

Any chance you tag a new beta release, please?

kismetwireless, 2019-01-02:
There will be a beta two at some point in the coming days or weeks, yes.

Beta 1 should compile fine, however

wrobelda, 2019-01-02:
> Beta 1 should compile fine, however

In that issue I referenced you say yourself that "The solution is to use the git-master code which does not include references to native thread handles (and hasn't since early august)".

closed 2019-07-02 by kismetwireless