kismetwireless/kismet #107: Could not import the KismetCaptureRtladsb Python code; you need to install  the rest of the rtladsb code from Kismet

WhistleMaster, 2019-01-03:
I followed the kismet setup guide to compile kismet fron github. Now, when starting kismet, I've got the following error on and on:

```Could not import the KismetCaptureRtladsb Python code; you need to install the rest of the rtladsb code from Kismet```

RPi 3b - Linux r02 4.14.62-Re4son-v7+ #2 SMP Sat Sep 22 22:53:40 AEST 2018 armv7l GNU/Linux

WhistleMaster, 2019-01-03:
Installing PyModeS fixed it.

closed 2019-01-03 by WhistleMaster

kismetwireless, 2019-01-03:
You installed part of the rtlasb code, but not, apparently, all of it - you
need to have the python module installed in your python directory.  Look
for errors while compiling or installing to see why this didn't happen;
it's done via python modules.

You will only get that message repeatedly if you keep adding a new source
or are in the source probe window.

dragorn, 2019-01-03:
i'll see about adding pymodes as a dep!