kismetwireless/kismet #108: Kismet DB abstraction

ashmastaflash, 2019-01-12:
Hi Kismet team,
First off, thank you SO MUCH for creating such a solid, useful utility.

I would like to contribute to the logging utilities part of the codebase. Before I begin, I would just like to get some thoughts (interested or nah) in what I would like to build.

Here's what I'm thinking:
1) _Python abstraction for Kismet database_: I see the same logic implemented over and over, and the individual integrations could benefit from a common abstraction. That can also serve as a compensating factor if the schema in the DB changes. Thoughts on whether or not this is something you're interested in, and where (in the codebase) it would be best to put such an abstraction? Or should that exist as a separate project with its own test suite, maybe published through PyPi?

2) _Filebeat wrapper for Kismet logging_. Filebeat has the ability to accept events from stdin (https://www.elastic.co/guide/en/beats/filebeat/current/filebeat-input-stdin.html) so if we wrap that functionality and pipe events from the Kismet DB, we can find an easier path to ELK than directly managing ElasticSearch interaction.

I look forward to your thoughts, and all apologies if this is the wrong forum for this type of question.

dragorn, 2019-01-12:
Those seem like very good ideas.  I have no complaints if you want to start
a github repo and I'll link it in the kismet docs?  Alternately, you're
welcome to send me pull reqs (or both - track your own repo and then send
pull reqs as you hit milestones if you like?) Right now all the python code
is in python_modules/ in the source tree.

Ultimately getting things in pypi would be excellent; that's the ultimate
end-goal for the existing kismet python libs too (kismetexternal, etc).

I'd begun de-duping some of the database work into a kismet log tools
library in python_modules/ but I haven't gotten too far with it yet.

If you want to chat realtime, swing by the discord channel (link's on the
website); myself and some other folk to do a lot of extracting from the
database are usually around, just remember it's a bit of an async medium so
it might take us a little bit to respond at first.

kismetwireless, 2019-06-12:
This is underway and Ash has been working on some awesome integration; closing ticket

closed 2019-06-12 by kismetwireless