kismetwireless/kismet #113: Kali package expects libsensors4, while only libsensors5 is available

strasharo, 2019-02-19:
Tried to install the Kali package on latest Kali 2019.1, running on a Raspberry Zero W and it fails to find the dependency libsensors4, because it looks like only libsensors5 is currently available in the repositories:

```
root@zero2:~# cat /etc/os-release 
PRETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2019.1"
VERSION_ID="2019.1"
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="https://www.kali.org/"
SUPPORT_URL="https://forums.kali.org/"
BUG_REPORT_URL="https://bugs.kali.org/"
root@zero2:~# 

root@zero2:~# aptitude install kismet2018
The following NEW packages will be installed:
  kismet-capture-linux-bluetooth{a} kismet-capture-linux-wifi{a} kismet-capture-nrf-mousejack{a} kismet-core{ab} kismet-logtools{a} kismet2018 librtlsdr0{a} python-kismetcapturertl433{a} python-kismetexternal{a} python-usb{a} 
0 packages upgraded, 10 newly installed, 0 to remove and 0 not upgraded.
Need to get 5,270 kB of archives. After unpacking 15.9 MB will be used.
The following packages have unmet dependencies:
 kismet-core : Depends: libsensors4 which is a virtual package and is not provided by any available package

The following actions will resolve these dependencies:

     Keep the following packages at their current version:
1)     kismet-core [Not Installed]                        
2)     kismet2018 [Not Installed]                         

Accept this solution? [Y/n/q/?] q
Abandoning all efforts to resolve these dependencies.
Abort.
root@zero2:~# dpkg -l | grep -i libsensors
ii  libsensors-config                    1:3.5.0-3                   all          lm-sensors configuration files
ii  libsensors4-dev:armel                1:3.5.0-3                   armel        lm-sensors development kit
ii  libsensors5:armel                    1:3.5.0-3                   armel        library to read temperature/voltage/fan sensors
root@zero2:~# aptitude search libsensors
p   libsensors-applet-plugin-dev                                                                                   - Create plugins for the 'sensors-applet' package                                                                          
p   libsensors-applet-plugin0                                                                                      - Library for plugins for the 'sensors-applet' package                                                                     
i A libsensors-config                                                                                              - lm-sensors configuration files                                                                                           
v   libsensors-dev                                                                                                 -                                                                                                                          
i   libsensors4-dev                                                                                                - lm-sensors development kit                                                                                               
i A libsensors5                                                                                                    - library to read temperature/voltage/fan sensors                                                                          
root@zero2:~#

root@zero2:~# cat /etc/apt/sources.list.d/kismet.list 
deb https://www.kismetwireless.net/repos/apt/git/kali rolling main
root@zero2:~# 

```

strasharo, 2019-02-19:
Works fine now, thanks a lot.

closed 2019-02-19 by strasharo

GemeenAapje, 2019-11-21:
I have the same problem. Kali upgraded from 2017 to 2019 and now wont boot.
Complains that libsensors4 cannot be found and also that texlive-base version isn't correct.

Any ideas how you fixed it?

strasharo, 2019-11-22:
@dragorn updated the package dependencies back then, I didn't do anything just pulled the latest package from the repo. What's not booting, the whole OS?