kismetwireless/kismet #118: Would it be possible to change the version-string of kismet releases?

rubyFeedback, 2019-02-28:
Hello.

Kismet uses a somewhat rare version scheme such as "kismet-2016-07-R1"
and similar. This makes it a bit difficult to properly parse it into its constituent
substrings.

I would like to suggest to adopt a different string format such as:

"kismet-2016.07.R1"

So in other words, to make use of "." rather than "-".

it is not a big deal, I only have to change a regex, but on the other hand
if possible it would be great if kismet could adopt this slight change. (It
also makes it a bit harder because of e. g. substrings appearing like
"R1", but the larger issue is that with only "-" it is hard to properly split
it up.)

kismetwireless, 2019-02-28:
I don't have any particularly strong opinions; I'll let this soak for a few
days and see if anyone else comments on it (I don't know if any of the
package maintainers watch this anyhow).

Corin-EU, 2019-03-27:
Why not make the version number the full date and include the day as well as the year and the month?

Please do not change the hyphen "-" to a dot "." as this would be a corruption of the ISO 8601 date format.  But what I would suggest is to simply drop the"-" and have no separator
which would then render an alternative ISO 8601 format date viz YYYYMMDD (eg 20160724). 

If multiple revisions are required then append the "dot R number" eg ".R1", ".R2" etc, with my preference for a lowercase "r" rather than an uppercase "R".

The probability of more than 9 revisions per day is incredibly small, so there is no need for a two digit number 01-99.

My suggestion would simplify the regular expression needed to parse the version, and note that if a dot separator were used to separate parts of the date then it would have to be escaped '\." making the expression more unwieldy.

2[0-9][0-9][0-9][0-1][0-9][1-3][0-9]\.R[0-9]

You can change the 2 to [2-N] where N is the further number of millennia that you expect to be writing versions of kismet.  I am not aware of any releases of kismet prior to the year  2000.

Corin-EU, 2019-03-27:
In the posting above, literal backslashes have been eaten.  In the regexp it needs to be
"backslash".R

DiscoveringWorld, 2020-04-06:
For me its change nothing  . or - on a split function. I dont understand how that can be more difficult using a -  You just have to count the number of index.