kismetwireless/kismet #124: Issue while compiling on macOS

retriple333, 2019-03-12:
Ello;

I have been compiling the Kismet the Github repo on my mac, however when I attempted to `make`, it errored out.

I was given this immediately after hitting enter, and I can't make sense of it. I have followed all of the steps correctly in the past, so it should be working properly.
`$ make
g++ -std=gnu++14 -Wall -g -I. -g -O2 -O3 -Qunused-arguments -thread -I/opt/local/include -I/opt/local/include -I/opt/local/include  -c util.cc -o util.cc.o
util.cc:977:5: error: no matching function for call to 'pthread_setname_np'
    pthread_setname_np(*thread, name.c_str());
    ^~~~~~~~~~~~~~~~~~
/usr/include/pthread.h:500:5: note: candidate function not viable: requires 1
      argument, but 2 were provided
int     pthread_setname_np(const char*);
        ^
util.cc:981:5: error: no matching function for call to 'pthread_setname_np'
    pthread_setname_np(thread.native_handle(), name.c_str());
    ^~~~~~~~~~~~~~~~~~
/usr/include/pthread.h:500:5: note: candidate function not viable: requires 1
      argument, but 2 were provided
int     pthread_setname_np(const char*);
        ^
2 errors generated.
make: *** [util.cc.o] Error 1
`

Thanks in advance!

retriple333, 2019-03-12:
also, make: *** [util.cc.o] Error 1

digitallyelite, 2019-07-05:
Are you still running into this issue? I've built it on my Mac without issue here.

DiscoveringWorld, 2020-04-06:
Maybe you should review the requirement for the build. Its seem something is missing here