kismetwireless/kismet #131: protobuf error on Mac 10.10.5

debragail, 2019-03-30:
`Debras-MacBook-Pro:kismet 0PHE1IA$ make
g++ -std=gnu++14 -Wall -g -I. -g -O2 -O3 -Qunused-arguments -pthread -I/usr/local/Cellar/protobuf/3.7.1/include -I/opt/local/include -I/opt/local/include  -c protobuf_cpp/kismet.pb.cc -o protobuf_cpp/kismet.pb.cc.o
In file included from protobuf_cpp/kismet.pb.cc:5:
protobuf_cpp/kismet.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
protobuf_cpp/kismet.pb.h:18:2: error: incompatible with your Protocol Buffer headers. Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
protobuf_cpp/kismet.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
protobuf_cpp/kismet.pb.h:1101:82: error: unknown class name 'true_type'; did you mean 'std::true_type'?
template <> struct is_proto_enum< ::KismetExternal::MsgbusMessage_MessageType> : ::google::protobuf::internal::true_type {};
                                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                                 std::true_type
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/type_traits:263:40: note: 'std::true_type' declared here
typedef integral_constant<bool, true>  true_type;
                                       ^
protobuf_cpp/kismet.pb.cc:54:52: error: 'Command' does not refer to a value
    GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(Command, command_),
                                                   ^
protobuf_cpp/kismet.pb.h:68:7: note: declared here
class Command : public ::google::protobuf::Message {
      ^
protobuf_cpp/kismet.pb.cc:54:61: error: use of undeclared identifier 'command_'
    GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(Command, command_),
                                                            ^
protobuf_cpp/kismet.pb.cc:55:52: error: 'Command' does not refer to a value
    GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(Command, seqno_),
                                                   ^
protobuf_cpp/kismet.pb.h:68:7: note: declared here
class Command : public ::google::protobuf::Message {
      ^
protobuf_cpp/kismet.pb.cc:55:61: error: use of undeclared identifier 'seqno_'
    GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(Command, seqno_),
                                                            ^
protobuf_cpp/kismet.pb.cc:56:52: error: 'Command' does not refer to a value
    GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(Command, content_),
                                                   ^
protobuf_cpp/kismet.pb.h:68:7: note: declared here
class Command : public ::google::protobuf::Message {
      ^
protobuf_cpp/kismet.pb.cc:56:61: error: use of undeclared identifier 'content_'; did you mean 'cgetent'?
    GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(Command, content_),
                                                            ^~~~~~~~
                                                            cgetent
/usr/include/stdlib.h:258:6: note: 'cgetent' declared here
int      cgetent(char **, char **, const char *);
         ^
protobuf_cpp/kismet.pb.cc:63:54: error: 'Command' does not refer to a value
      GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(Command, _has_bits_[0]),
                                                     ^
protobuf_cpp/kismet.pb.h:68:7: note: declared here
class Command : public ::google::protobuf::Message {
      ^
protobuf_cpp/kismet.pb.cc:63:63: error: use of undeclared identifier '_has_bits_'
      GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(Command, _has_bits_[0]),
                                                              ^
protobuf_cpp/kismet.pb.cc:64:54: error: 'Command' does not refer to a value
      GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(Command, _unknown_fields_),
                                                     ^
protobuf_cpp/kismet.pb.h:68:7: note: declared here
class Command : public ::google::protobuf::Message {
      ^
protobuf_cpp/kismet.pb.cc:64:63: error: use of undeclared identifier '_unknown_fields_'
      GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(Command, _unknown_fields_),
                                                              ^
protobuf_cpp/kismet.pb.cc:71:52: error: 'MsgbusMessage' does not refer to a value
    GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(MsgbusMessage, msgtype_),
                                                   ^
protobuf_cpp/kismet.pb.h:177:7: note: declared here
class MsgbusMessage : public ::google::protobuf::Message {
      ^
protobuf_cpp/kismet.pb.cc:71:67: error: use of undeclared identifier 'msgtype_'
    GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(MsgbusMessage, msgtype_),
                                                                  ^
protobuf_cpp/kismet.pb.cc:72:52: error: 'MsgbusMessage' does not refer to a value
    GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(MsgbusMessage, msgtext_),
                                                   ^
protobuf_cpp/kismet.pb.h:177:7: note: declared here
class MsgbusMessage : public ::google::protobuf::Message {
      ^
protobuf_cpp/kismet.pb.cc:72:67: error: use of undeclared identifier 'msgtext_'
    GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(MsgbusMessage, msgtext_),
                                                                  ^
protobuf_cpp/kismet.pb.cc:79:54: error: 'MsgbusMessage' does not refer to a value
      GOOGLE_PROTOBUF_GENERATED_MESSAGE_FIELD_OFFSET(MsgbusMessage, _has_bits_[0]),
                                                     ^
protobuf_cpp/kismet.pb.h:177:7: note: declared here
class MsgbusMessage : public ::google::protobuf::Message {
      ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make: *** [protobuf_cpp/kismet.pb.cc.o] Error 1
`

kismetwireless, 2019-03-30:
It looks like you've upgraded your version of protobuf; you need to
regenerate the protobuf code in Kismet.

make distclean and run configure again and it should sort it out; or use a
fresh checkout of Kismet.

debragail, 2019-04-01:
`Debras-MacBook-Pro:kismet 0PHE1IA$ make distclean
# @-rm -f kaitai_parsers/*.o
# @-rm -f kaitai_parsers/*.d
PLUGIN-CLEAN: plugin-alertsyslog/
PLUGIN-CLEAN: plugin-dashboard/
Nothing to clean
PLUGIN-CLEAN: plugin-demo-webonly/
Nothing to clean
PLUGIN-CLEAN: plugin-httpproxytest/
running clean
running clean
running clean
PLUGIN-CLEAN: plugin-alertsyslog/
PLUGIN-CLEAN: plugin-dashboard/
Nothing to clean
PLUGIN-CLEAN: plugin-demo-webonly/
Nothing to clean
PLUGIN-CLEAN: plugin-httpproxytest/
Debras-MacBook-Pro:kismet 0PHE1IA$ ./configure
checking build system type... x86_64-apple-darwin14.5.0
checking host system type... x86_64-apple-darwin14.5.0
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for a BSD-compatible install... /usr/local/bin/ginstall -c
checking whether make sets $(MAKE)... yes
checking how to run the C preprocessor... gcc -E
checking for platform-specific compiler flags... Apple GCC - added Apple80211 frameworks and no-precomp
checking gcc version... 4.2.1
checking whether g++ supports C++17 features with -std=gnu++17... no
checking whether g++ supports C++17 features with -std=gnu++1z... no
checking whether g++ supports C++17 features with -std=c++17... no
checking whether g++ supports C++17 features with +std=c++17... no
checking whether g++ supports C++17 features with -h std=c++17... no
checking whether g++ supports C++17 features with -std=c++1z... no
checking whether g++ supports C++17 features with +std=c++1z... no
checking whether g++ supports C++17 features with -h std=c++1z... no
configure: No compiler with C++17 support was found
checking whether g++ supports C++14 features with -std=gnu++14... yes
checking how to run the C++ preprocessor... g++ -std=gnu++14 -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking whether byte ordering is bigendian... no
checking python module: setuptools... yes
checking for dwarf_begin in -ldw... no
checking for bfd_alloc in -lbfd... no
checking unwind.h usability... yes
checking unwind.h presence... yes
checking for unwind.h... yes
checking execinfo.h usability... yes
checking execinfo.h presence... yes
checking for execinfo.h... yes
checking for stdint.h... (cached) yes
checking for accept() addrlen type... socklen_t
checking for pipe2... NO
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking for dlopen in -ldl... yes
checking for deflate in -lz... yes
checking for pthread_mutex_timedlock... no
checking for __gnu_parallel extensions... no
checking for libm math function in std libs... yes
checking for main in -lstdc++... yes
checking for group 'root'... no.  Using 'wheel'
checking for MHD_start_daemon in -lmicrohttpd... yes
checking microhttpd.h usability... yes
checking microhttpd.h presence... yes
checking for microhttpd.h... yes
checking for setproctitle... no
checking libutil.h usability... no
checking libutil.h presence... no
checking for libutil.h... no
checking for setproctitle in -lutil... no
checking sys/pstat.h usability... no
checking sys/pstat.h presence... no
checking for sys/pstat.h... no
checking whether __progname and __progname_full are available... no
checking which argv replacement method to use... writeable
checking for cap_init in -lcap... no
checking sys/prctl.h usability... no
checking sys/prctl.h presence... no
checking for sys/prctl.h... no
checking sys/capability.h usability... no
checking sys/capability.h presence... no
checking for sys/capability.h... no
checking for pcre_compile in -lpcre... yes
checking pcre.h usability... yes
checking pcre.h presence... yes
checking for pcre.h... yes
checking for sqlite3_libversion in -lsqlite3... yes
checking sqlite3.h usability... yes
checking sqlite3.h presence... yes
checking for sqlite3.h... yes
checking for pcap_open_live in -lpcap... yes
checking pcap.h usability... yes
checking pcap.h presence... yes
checking for pcap.h... yes
checking for pcap_setnonblock in -lpcap... yes
checking for pcap_get_selectable_fd in -lpcap... yes
checking for PPI support in libpcap... yes
checking for pkg-config... /usr/local/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for protobuf... yes
checking for protoc... yes
checking for libprotobufc... no
checking google/protobuf-c/protobuf-c.h usability... no
checking google/protobuf-c/protobuf-c.h presence... no
checking for google/protobuf-c/protobuf-c.h... no
checking for protobuf_c_message_pack_to_buffer in -lprotobuf-c... no
configure: error: missing libprotobuf-c`

kismetwireless, 2019-04-01:
You're missing part of protobuf in your host install.

debragail, 2019-04-01:
i just brew installed protobuf-c again

was able to ./configure and now on sudo make suidinstall it fails with this:

`gcc -Wall -g -I.    -pthread -I/usr/local/Cellar/protobuf-c/1.3.1_2/include  -c version.c -o version.c.o
g++ -std=gnu++14 -framework Foundation -framework CoreFoundation -F/System/Library/PrivateFrameworks -framework Apple80211 -framework IOKit -rdynamic -o kismet util.cc.o macaddr.cc.o uuid.cc.o xxhash.cc.o boost_like_hash.cc.o sqlite3_cpp11.cc.o globalregistry.cc.o eventbus.cc.o pollabletracker.cc.o ringbuf2.cc.o chainbuf.cc.o filewritebuf.cc.o buffer_handler.cc.o packet.cc.o messagebus.cc.o configfile.cc.o getopt.cc.o psutils.cc.o battery.cc.o tcpserver2.cc.o tcpclient2.cc.o serialclient2.cc.o pipeclient.cc.o ipc_remote2.cc.o protobuf_cpp/kismet.pb.cc.o protobuf_cpp/http.pb.cc.o protobuf_cpp/datasource.pb.cc.o protobuf_cpp/linuxbluetooth.pb.cc.o  kis_external.cc.o dlttracker.cc.o antennatracker.cc.o datasourcetracker.cc.o kis_datasource.cc.o datasource_linux_bluetooth.cc.o datasource_rtl433.cc.o datasource_rtlamr.cc.o datasource_rtladsb.cc.o kis_net_microhttpd.cc.o system_monitor.cc.o base64.cc.o kis_httpd_websession.cc.o kis_httpd_registry.cc.o gpstracker.cc.o kis_gps.cc.o gpsnmea.cc.o gpsserial2.cc.o gpstcp.cc.o gpsgpsd2.cc.o gpsfake.cc.o gpsweb.cc.o packetchain.cc.o packet_filter.cc.o class_filter.cc.o trackedelement.cc.o trackedcomponent.cc.o entrytracker.cc.o trackedlocation.cc.o devicetracker_component.cc.o devicetracker_view.cc.o devicetracker_view_workers.cc.o jsoncpp.cc.o json_adapter.cc.o plugintracker.cc.o alertracker.cc.o timetracker.cc.o channeltracker2.cc.o devicetracker.cc.o devicetracker_workers.cc.o devicetracker_httpd.cc.o kis_dlt.cc.o kis_dlt_ppi.cc.o kis_dlt_radiotap.cc.o kaitaistream.cc.o dot11_parsers/dot11_ie.cc.o dot11_parsers/dot11_ie_7_country.cc.o dot11_parsers/dot11_ie_11_qbss.cc.o dot11_parsers/dot11_ie_33_power.cc.o dot11_parsers/dot11_ie_36_supported_channels.cc.o dot11_parsers/dot11_ie_45_ht_cap.cc.o dot11_parsers/dot11_ie_48_rsn.cc.o dot11_parsers/dot11_ie_52_rmm_neighbor.cc.o dot11_parsers/dot11_ie_54_mobility.cc.o dot11_parsers/dot11_ie_55_fastbss.cc.o dot11_parsers/dot11_ie_61_ht_op.cc.o dot11_parsers/dot11_ie_70_rm_capabilities.cc.o dot11_parsers/dot11_ie_133_cisco_ccx.cc.o dot11_parsers/dot11_ie_150_vendor.cc.o dot11_parsers/dot11_ie_150_cisco_powerlevel.cc.o dot11_parsers/dot11_ie_191_vht_cap.cc.o dot11_parsers/dot11_ie_192_vht_op.cc.o dot11_parsers/dot11_ie_127_extended_capabilities.cc.o dot11_parsers/dot11_ie_221_vendor.cc.o dot11_parsers/dot11_ie_221_ms_wps.cc.o dot11_parsers/dot11_ie_221_ms_wmm.cc.o dot11_parsers/dot11_ie_221_dji_droneid.cc.o dot11_parsers/dot11_ie_221_wfa_wpa.cc.o dot11_parsers/dot11_ie_221_cisco_client_mfp.cc.o dot11_parsers/dot11_action.cc.o dot11_parsers/dot11_wpa_eap.cc.o phy_80211.cc.o phy_80211_dissectors.cc.o phy_rtl433.cc.o phy_rtlamr.cc.o phy_rtladsb.cc.o phy_zwave.cc.o phy_bluetooth.cc.o phy_uav_drone.cc.o phy_nrf_mousejack.cc.o dot11_fingerprint.cc.o kis_dissector_ipdata.cc.o manuf.cc.o logtracker.cc.o kis_ppilogfile.cc.o kis_databaselogfile.cc.o kis_pcapnglogfile.cc.o messagebus_restclient.cc.o streamtracker.cc.o pcapng_stream_ringbuf.cc.o streambuf_stream_buffer.cc.o devicetracker_httpd_pcap.cc.o phy_80211_httpd_pcap.cc.o kis_database.cc.o storageloader.cc.o kismet_server.cc.o  version.c.o -ldl -lz -lpcre -lsqlite3  -lpcap -lmicrohttpd -lpthread -L/usr/local/Cellar/protobuf/3.7.1/lib -lprotobuf -Qunused-arguments -pthread -rdynamic
Undefined symbols for architecture x86_64:
  "std::__1::shared_timed_mutex::unlock_shared()", referenced from:
      Datasourcetracker::find_datasource(uuid const&) in datasourcetracker.cc.o
      Datasourcetracker::Httpd_VerifyPath(char const*, char const*) in datasourcetracker.cc.o
      Datasourcetracker::Httpd_CreateStreamResponse(Kis_Net_Httpd*, Kis_Net_Httpd_Connection*, char const*, char const*, char const*, unsigned long*, std::__1::basic_stringstream<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in datasourcetracker.cc.o
      Datasourcetracker::Httpd_PostComplete(Kis_Net_Httpd_Connection*) in datasourcetracker.cc.o
      DevicetrackerView::doDeviceWork(DevicetrackerViewWorker&) in devicetracker_view.cc.o
      DevicetrackerView::doReadonlyDeviceWork(DevicetrackerViewWorker&) in devicetracker_view.cc.o
      DevicetrackerView::doReadonlyDeviceWork(DevicetrackerViewWorker&, std::__1::shared_ptr<TrackerElementVector>) in devicetracker_view.cc.o
      ...
  "std::__1::shared_timed_mutex::lock()", referenced from:
      Kis_Net_Httpd_Buffer_Stream_Handler::buffer_event_cb(void*, unsigned long long, char*, unsigned long) in kis_net_microhttpd.cc.o
      free_buffer_aux_callback(void*) in kis_net_microhttpd.cc.o
  "std::__1::shared_timed_mutex::unlock()", referenced from:
      GlobalRegistry::RegisterGlobal(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in globalregistry.cc.o
      GlobalRegistry::FetchGlobalRef(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in globalregistry.cc.o
      GlobalRegistry::FetchGlobal(int) in globalregistry.cc.o
      GlobalRegistry::FetchGlobal(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in globalregistry.cc.o
      GlobalRegistry::InsertGlobal(int, std::__1::shared_ptr<void>) in globalregistry.cc.o
      GlobalRegistry::RemoveGlobal(int) in globalregistry.cc.o
[... cut ...]
      PollableTracker::PollableTracker(GlobalRegistry*) in pollabletracker.cc.o
      RingbufV2::RingbufV2(unsigned long) in ringbuf2.cc.o
      Chainbuf::Chainbuf(unsigned long, unsigned long) in chainbuf.cc.o
      FileWritebuf::FileWritebuf(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, unsigned long) in filewritebuf.cc.o
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [kismet] Error 1`

kismetwireless, 2019-04-01:
Unfortunately I don't have access to the latest OSX so I have no further
suggestions; the latest git compiles fine & as expected under high sierra
though.

wtarreau, 2024-11-13:
same issue here on linux (protobuf-c-3.0.18, only version available), the protobuf stuff seems very brittle. Distclean didn't change anything and I have no idea how to regenerate such files. The procedure to rebuild them should probably be documented in the readme I guess, because I suspect it should not be that hard. But when you're totally ignorant of this susceptible library it's impossible to go further.

wtarreau, 2024-11-13:
Looks like the last buildable commit for me is from 2018: c77c99c74cd