kismetwireless/kismet #132: kali apt repository: "expected kali but got rolling"

TechInterMezzo, 2019-03-31:
I get the following error when I try to install kismet with apt on raspberry pi kali.

`W: Conflicting distribution: https://www.kismetwireless.net/repos/apt/release/ka                                           li kali InRelease (expected kali but got rolling)`

If I look directly at https://www.kismetwireless.net/repos/apt/release/kali/dists/kali/InRelease,
it says "Codename: rolling" at line 6.

The expected name would be kali instead of rolling.

kismetwireless, 2019-03-31:
Change your apt line to 'rolling' for the releases repo for now; it will be
replaced with 'kali' with the next release build when beta3 comes out,
however.

kismetwireless, 2019-03-31:
I'll work on rebuilding the release repos tonight maybe which will likely
make the error go away anyhow

kismetwireless, 2019-03-31:
The release repos are now rebuild with aptly, which changed some things;
should now match the codename 'kali'

dragorn, 2019-03-31:
You may need to pull the nightly repos depending on your kali version
still; kali recently swapped some libraries and upgraded them & those
aren't in the beta2 builds.  beta3 is relatively immanent though.

closed 2019-04-02 by TechInterMezzo