kismetwireless/kismet #135: Kismet 2019-04-R1 source release tarball leads to sources failing followed by impossibility to ctrl+c out

stephantual, 2019-04-07:
Kismet 2019-04-R1 source release tarball
built from source on raspi 3 b+
Kali latest + apt-get upgrade done

While Kismet 2019-02 was running fine, Kismet 2019-04-R1 sees both of my sources (ALFA AWUS036ACH and ALFA AWUS036NHA) 'failing' then 'retrying' over and over again. 

Example would be: 
"data source wlan2:name=AWUS036NHA" launched successfully. 
... then a few detections as expected,. followed by:
"data source wlan2:name=AWUS036NHA"(wlan2) encounered an error: pipe closing - remote side closed pipe|
... followed by a few retries and eventually:
" data source 'wlan2:name=AWUS036NHA" launching while an IPC process is already running : killing existing process 12489"
... ad nauseam.

I then try to kill kismet with ctrl+c to terminate it as usual, but it's not providing any other feedback, just hangs until a kill-9 or a reboot. 

I can 'vacuum' the journals to 'salvage' the bits and bobs captured before the first failures of the sources but it's evidently not-ideal, especially as I couldn't find more informaion or trace from the " IPC process is already running : killing existing process" error. 

Clues? I'm happy to help by debugging further.

stephantual, 2019-04-13:
Ok , more about this - I've then installed dietpi with just raspian on it so there would be no doubt that the distro was not at the core of the issue. For the atheros chipset i had no issue, works out of the box. For the realtek chipset i thankfully had the headers and compiled the drivers based on the aircrack version. The box is as light, steady and clear of junk as you can think of.

So i get this power adapter dedicated to the pi from raspberry themselves in case it's a matter of power, and i do everything by the book. Let kismet source release tarball run overnight... and the exact same error occurs. 

What's really frustrating is that I can't find where kismet is storing its error logs. I need to know why the cards fail if i want to have a chance to fix it. That or it's the release, which i'll have to test by reverting to the old version but it's starting to be quite a bit of work!

tgies, 2019-05-02:
Not a solution, but my experience with a similar issue where Kismet appears to hang and spew messages about connection failures suggests that if you press Ctrl-C and wait several minutes it will eventually clean up and terminate normally.

FWIW I am using source devices with the same chipsets as AWUS036ACH (ath9k) and AWUS036NHA (rtl8812au) and they work fine on Kali armhf and arm64 rolling on a Pi 3 B here. I am using [the official nightly git packages](https://kismetwireless.net/docs/readme/packages/).

Something to try: Instead of starting your sources from within Kismet, start the capture program yourself  at the terminal (`kismet_cap_linux_wifi --connect localhost:3501 --source wlan2`) -- do you see the capture program die or log errors or anything?

kismetwireless, 2019-06-12:
This looks like a set of bugs that are almost entirely resolved in 2019-05-02, with the last few of them resolved in 2019-06-git which will become the next release.

closed 2019-06-12 by kismetwireless