kismetwireless/kismet #139: Kismet hangs with GPSd configured but not connectable

tgies, 2019-05-01:
If Kismet is configured to use GPSd, but the configured GPSd is not connectable, the Kismet server will enter a weird pseudo-hang state after an unpredictable period of time.

When Kismet is configured to use a GPSd that is not running, it will periodically log messages about reattempting connection to GPSd that look like this:

    ERROR: GPSDv2 didn't get data from gpsd in over 30 seconds, reconnecting
           to GPSD server.

Eventually, immediately after logging one of these messages, Kismet will enter the hang state.

In this state:
- Parts of the WebUI become unresponsive,
- Kismet will periodically log messages about connections to sources failing, and/or more messages about connections to GPSd failing.
- Kismet logs no packets (rarely it may log one or two here and there, but if you are watching the WebUI the packet graphs will distinctly "flatline" at the moment it hangs),
- On pressing Ctrl-C, Kismet appears not to respond, but will eventually terminate after a long time (potentially 10+ minutes).

I have reproduced this on:
- Kali rolling arm64 on Raspberry Pi 3 B
- Kali rolling armhf on Raspberry Pi 3 B
- Ubuntu 18.04 on x86_64

Steps to reproduce:
- Configure Kismet with the following GPS source line in `kismet.conf`, and no other GPS source:
    `gps=gpsd:host=localhost,port=2947`
- Ensure no GPSd is running locally and there is no other listener on 127.0.0.1:2947
- Start Kismet and enable at least one source.
- Wait. In some cases the problem takes over 20 minutes to manifest.

Reproduced on `86d7672`.

kismetwireless, 2019-05-01:
Currently I can't reproduce this on any hardware, over days of running w/
an unavailable gpsd.  I'll continue testing in the meantime, but it's been
running for literally a month on one server.

dragorn, 2019-05-01:
Pushed some more semi-blind changes; found one possible thread race between
clearing the buffer handler and the thread-0 poll loop; no idea if it was
triggering but it's a shot.

kismetwireless, 2019-05-02:
OK; I did get it to act strangely, finally, and attached gdb.  I just
pushed another set of GPS tweaks; they may help.  I haven't been able to
get it to happen again, anyhow.

tgies, 2019-05-02:
Thanks, will try these changes.

Just wanted to add, also, that the one system where I *can't* yet repro this is OpenWRT 18.06.1 running on an ar71xx device (QCA9563), built using OpenWRT SDK and your package definition files. This is also the only non-multiprocessing system I've tried, and I'm not sure if that's relevant, but it's the only variable I can identify that might be.

At some point between e2ff432 and 0ec810a, it seems like this changed from an event that occurred reliably on the 3rd-4th loop of reconnecting to GPSd to something that can take 20+ minutes to happen. I think the connection/buffer error handling changes made around there mitigated this problem somewhat. I can try a bisection around there if you're interested in finding out what affected it, but I will be trying the latest changes you pushed first.

Just to get this in the public issue thread, I'm also having an issue (that you know about from our previous email discussion) where Kismet eventually hangs in much the same way *with* GPSd connected, but I'm focusing on this issue because it's easier to repro (here on my machines, anyway) and I suspect it to be a manifestation of the same underlying bug.

kismetwireless, 2019-06-12:
LIke we discussed on Discord, I think we narrowed this down and it ought to be fixed for the most part in 2019-05-02 and finally in the 2019-06-git, so I'll close this up.

closed 2019-06-12 by kismetwireless