kismetwireless/kismet #144: Crash at launch

limburgher, 2019-05-08:
I'm the Fedora kismet package maintainer. I've found that both 2019-04-R1 and master crash at launch. See attached log.
[log.txt](https://github.com/kismetwireless/kismet/files/3159278/log.txt)

kismetwireless, 2019-05-09:
#10   Source "/home/gwyn/dev/kismet/kismet_server.cc", line 912, in _start
        909:         if (globalregistry->fatal_condition) {
        910:             _MSG_FATAL("Failure activating Kismet plugins,
make sure that all your plugins "
        911:                     "are built against the same version of
Kismet.");
      > 912:             SpindownKismet(pollabletracker);
        913:         }
        914:     }

Looks like you've got bits of old plugins installed on your test system;
alternately, you haven't got things properly installed or are using ancient
config files, because also:

ERROR: No httpd_home defined in kismet.conf, disabling static file
       serving. This will disable the web UI, but the REST interface will
       still function.
ERROR: No httpd_user_home defined in kismet.conf, disabling static file
       serving from user directory

Those alone will render Kismet basically unusable.

To narrow it down more I'd need to see a debug log, or at the least, a lot
with --disable-ncurses so that the complete history is available.  Info
about running Kismet under a debugger is available at:

https://www.kismetwireless.net/docs/readme/debugging/

limburgher, 2019-05-09:
It was the old configs. Thank you. :)

closed 2019-05-09 by limburgher

kismetwireless, 2019-05-09:
Awesome.  If you get a chance to crash it again w/ a log I'll take a look;
otherwise I'll try to do the same myself; it should error, not crash, so
I'm not handling some situation properly; glad to know that fixed it tho!

dragorn, 2019-05-10:
I was unregistering the http handler before spinning down the webserver
core; that might have been causing the abort.