kismetwireless/kismet #145: 2019.04.R1 bug

xosevp, 2019-05-09:
runnig "kismet -c wlan0" in Fedora-30 I get:

/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = char; _Alloc = std::allocator<char>; std::vector<_Tp, _Alloc>::reference = char&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Stack trace (most recent call last) in thread 2444:
#13   Object "[0xffffffffffffffff], at 0xffffffffffffffff, in 
#12   Object "/lib64/libc.so.6, at 0x7ff132d94162, in clone
#11   Object "/lib64/libpthread.so.0, at 0x7ff1335255a1, in 
#10   Object "/lib64/libstdc++.so.6, at 0x7ff1330d1473, in 
#9    Object "kismet, at 0x692fcc, in Packetchain::packet_queue_processor()
#8    Object "kismet, at 0x7aae36, in Kis_80211_Phy::CommonClassifierDot11(GlobalRegistry*, void*, kis_packet*)
#7    Object "kismet, at 0x7a41d5, in Kis_80211_Phy::HandleSSID(std::shared_ptr<kis_tracked_device_base>, std::shared_ptr<dot11_tracked_device>, kis_packet*, dot11_packinfo*, kis_gps_packinfo*)
#6    Object "kismet, at 0x7ed051, in Kis_80211_Phy::PacketDot11IEdissector(kis_packet*, dot11_packinfo*)
#5    Object "kismet, at 0x786f72, in dot11_ie::parse(std::shared_ptr<kaitai::kstream>)
#4    Object "kismet, at 0x786bbf, in dot11_ie::dot11_ie_tag::parse(std::shared_ptr<kaitai::kstream>)
#3    Object "kismet, at 0x7856be, in kaitai::kstream::read_bytes[abi:cxx11](long)
#2    Object "kismet, at 0x5719a7, in std::__replacement_assert(char const*, int, char const*, char const*)
#1    Object "/lib64/libc.so.6, at 0x7ff132cbb894, in abort
#0    Object "/lib64/libc.so.6, at 0x7ff132cd0eb5, in gsignal
Aborted (Signal sent by tkill() 2441 0)
Aborted (core dumped)
FATAL:  Error during write(): Broken pipe

kismetwireless, 2019-05-09:
Please run it in debug mode from the following directions and post the log,
and I'll take a look:

https://www.kismetwireless.net/docs/readme/debugging/

dragorn, 2019-05-09:
It looks like from the trace you posted though that it's an uncaught
exception, rather than an outright crash - this is good.

I found a few things in the IE parsers that were parsing tags without a
try/catch around them, so I'll be pushing a fix to that to git soon, too,
so if you want to try the latest git code later today it might be addressed.

xosevp, 2019-05-11:
With current git code I get: https://paste.fedoraproject.org/paste/-gQHXRtdNXOms52xrqv4Ag

Do you need more info?

kismetwireless, 2019-05-11:
It tells me more, but isn't making much sense.

I just merged the latest kaitai runtime code which Kismet uses the stream
functions of; if it still crashes in there, could you both:

1) Run it with the debug commands from here and send me the log from gdb:
https://www.kismetwireless.net/docs/readme/debugging/
2) Fire up tcpdump and capture the packets at the same time Kismet is
running (launching tcpdump before kismet starts should be fine); something
like:
   tcpdump -i whateve0mon -s 0 -w debug.pcap

launch it before you run kismet and let it run until kismet crashes; feel
free to email me the pcap privately if you'd like.

You might need to run kismet once first to get it to make the monitor
interface for you.