kismetwireless/kismet #150: Kismet out of memory raspberry pi 3

arifainchtein, 2019-06-18:
Hello,
I installed kismet on the pi 3 using the file:

wget http://kismetwireless.net/code/kismet-2016-07-R1.tar.xz

and i got it to work ok. I am using a usb dongle with the 5370 chipset.
 The problem was that i need to export the data into json and it seems that the version i installed does not support that. 
I decided to install a release version following the instructions in :
https://www.kismetwireless.net/docs/readme/packages/

The problem now is that when i start kismet it quits immediately, and the only error i been able to see is:

kismetwireless, 2019-06-18:
That version is pretty ancient and unsupported. I'd suggest the latest
stable release, 2019.05.02.

It doesn't surprise me the old code doesn't work; the drivers have evolved
over time.

The pi3 isn't an optimal platform but you'll have some success.

arifainchtein, 2019-06-18:
i accidentally clicked the submit message before fuinishing the post.

i am trying to get the latest release to work, but It just dies with that out of memory message
Is there something i can do?  Would loe to be able to run it in the pi, because i am creating an embedded system.

thanks

kismetwireless, 2019-06-18:
Stable should handle 15000 to 30000 devices on a pi3.

Compiling it is another matter; you'll either need a lot of swap and a lot
of time, or use one of the packages that's precompiled.

You'll also need to not be running a desktop environment, etc on the pi if
you expect usable performance.

arifainchtein, 2019-06-18:
i just installed it, so no compiling.  Any ideas why I cant get it to start?  Any changes to kismet_memory.conf i should make?

thanks

arifainchtein, 2019-06-18:
oh, yes, i forgot to mention that the pi is running headless.

digitallyelite, 2019-07-02:
Are you attempting to run it from /etc/kismet ?  Your screenshot makes it seem so - the pi user can't write to that directory.

MyProjectExpert, 2020-01-14:
I have similar problem when running Kismet for several days.  Memory usage grows and eventually, the swap file fills and systems gets terribly slow.   If you run with --no-logging, this will help allow kismet to run much longer. 
Also, I haven't tested this yet, but configuring logging.conf to create new log files may help with memory as well

kismetwireless, 2020-01-14:
The more kismet sees, the more ram it needs - so this is expected.

You can tune kismets memory use:

https://www.kismetwireless.net/docs/readme/performance_and_memory/

and should definitely do so if you're attempting to run it on very small hw
like a pi3.