kismetwireless/kismet #152: C99 variable declaration in capture_nrf_mousejack

thundercleese80, 2019-06-30:
CentOS 7.6.1810
gcc-4.8.5-36.el7_6.2.x86_64
make-3.82-23.el7.x86_64

Pulled from git, kismet-2019-04-R1-249-g42eee84, no further modifications.

I get this when doing `make`:

gcc -Wall -g -I. -pthread -I/usr/include/libnm -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include   -I/usr/include/libnl3    -fopenmp -pthread   -fopenmp  -I/usr/include/libusb-1.0   -c capture_nrf_mousejack.c -o capture_nrf_mousejack.c.o
capture_nrf_mousejack.c: In function ‘probe_callback’:
capture_nrf_mousejack.c:239:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (ssize_t i = 0; i < libusb_devices_cnt; i++) {
     ^
capture_nrf_mousejack.c:239:5: note: use option -std=c99 or -std=gnu99 to compile your code
capture_nrf_mousejack.c:281:14: error: conflicting types for ‘i’
     for (int i = 2; i < 84; i++) {
              ^
capture_nrf_mousejack.c:239:18: note: previous definition of ‘i’ was here
     for (ssize_t i = 0; i < libusb_devices_cnt; i++) {
                  ^
capture_nrf_mousejack.c:281:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int i = 2; i < 84; i++) {
     ^
capture_nrf_mousejack.c:204:9: warning: variable ‘matched_device’ set but not used [-Wunused-but-set-variable]
     int matched_device = 0;
         ^
capture_nrf_mousejack.c: In function ‘list_callback’:
capture_nrf_mousejack.c:319:18: error: conflicting types for ‘i’
     for (ssize_t i = 0; i < libusb_devices_cnt; i++) {
                  ^
capture_nrf_mousejack.c:309:18: note: previous declaration of ‘i’ was here
     unsigned int i;
                  ^
capture_nrf_mousejack.c:319:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (ssize_t i = 0; i < libusb_devices_cnt; i++) {
     ^
capture_nrf_mousejack.c: In function ‘open_callback’:
capture_nrf_mousejack.c:429:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (ssize_t i = 0; i < libusb_devices_cnt; i++) {
     ^
capture_nrf_mousejack.c:485:14: error: conflicting types for ‘i’
     for (int i = 2; i < 84; i++) {
              ^
capture_nrf_mousejack.c:429:18: note: previous definition of ‘i’ was here
     for (ssize_t i = 0; i < libusb_devices_cnt; i++) {
                  ^
capture_nrf_mousejack.c:485:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int i = 2; i < 84; i++) {
     ^
make[1]: *** [capture_nrf_mousejack.c.o] Error 1
make[1]: Leaving directory `/home/brianraker/git/kismet/capture_nrf_mousejack'
make: *** [capture_nrf_mousejack/kismet_cap_nrf_mousejack] Error 2

thundercleese80, 2019-06-30:
Sorry, should also say master branch.

kismetwireless, 2019-06-30:
Weird config for your compiler, i haven't seen any other modern - or even not so modern - gcc or clang that doesn't support c99 properly out of the box, back to ubuntu 14...

Anyhow, configure will now explicitly force and require c99 format so hopefully that addresses it.