kismetwireless/kismet #154: Error while - sudo make && sudo make plugins - sudo make && sudo make plugins

aqua95, 2019-07-01:
sudo make && sudo make plugins
g++ -std=gnu++17 -MM -Wall -g -I. -g -O2 -O3 -pthread -DKS_STR_ENCODING_NONE -fopenmp  -MT kis_ppilogfile.cc.o kis_ppilogfile.cc > kis_ppilogfile.cc.d
g++ -std=gnu++17 -Wall -g -I. -g -O2 -O3 -pthread -DKS_STR_ENCODING_NONE -fopenmp  -c kis_ppilogfile.cc -o kis_ppilogfile.cc.o
kis_ppilogfile.cc:30:1: error: ‘KisPPILogfile’ does not name a type
 KisPPILogfile::KisPPILogfile(SharedLogBuilder in_builder) :
 ^~~~~~~~~~~~~
kis_ppilogfile.cc:54:6: error: ‘KisPPILogfile’ has not been declared
 bool KisPPILogfile::Log_Open(std::string in_path) {
      ^~~~~~~~~~~~~
kis_ppilogfile.cc: In function ‘bool Log_Open(std::__cxx11::string)’:
kis_ppilogfile.cc:55:24: error: ‘log_mutex’ was not declared in this scope
     local_locker lock(&log_mutex);
                        ^~~~~~~~~
kis_ppilogfile.cc:57:5: error: ‘set_int_log_path’ was not declared in this scope
     set_int_log_path(in_path);
     ^~~~~~~~~~~~~~~~
kis_ppilogfile.cc:59:2: error: ‘dumpfile’ was not declared in this scope
  dumpfile = NULL;
  ^~~~~~~~
kis_ppilogfile.cc:59:2: note: suggested alternative: ‘tmpfile’
  dumpfile = NULL;
  ^~~~~~~~
  tmpfile
kis_ppilogfile.cc:60:2: error: ‘dumper’ was not declared in this scope
  dumper = NULL;
  ^~~~~~
kis_ppilogfile.cc:60:2: note: suggested alternative: ‘dup3’
  dumper = NULL;
  ^~~~~~
  dup3
kis_ppilogfile.cc:61:5: error: ‘dump_filep’ was not declared in this scope
     dump_filep = NULL;
     ^~~~~~~~~~
kis_ppilogfile.cc:61:5: note: suggested alternative: ‘tmpfile’
     dump_filep = NULL;
     ^~~~~~~~~~
     tmpfile
kis_ppilogfile.cc:66:13: error: ‘pcap_open_dead’ was not declared in this scope
  dumpfile = pcap_open_dead(DLT_PPI, MAX_PACKET_LEN);
             ^~~~~~~~~~~~~~
kis_ppilogfile.cc:85:11: error: ‘pcap_dump_fopen’ was not declared in this scope
  dumper = pcap_dump_fopen(dumpfile, dump_filep);
           ^~~~~~~~~~~~~~~
kis_ppilogfile.cc:95:5: error: ‘set_int_log_open’ was not declared in this scope
     set_int_log_open(true);
     ^~~~~~~~~~~~~~~~
kis_ppilogfile.cc:95:5: note: suggested alternative: ‘sqlite3_blob_open’
     set_int_log_open(true);
     ^~~~~~~~~~~~~~~~
     sqlite3_blob_open
kis_ppilogfile.cc:97:32: error: ‘KisPPILogfile’ has not been declared
  packetchain->RegisterHandler(&KisPPILogfile::packet_handler, this, CHAINPOS_LOGGING, -100);
                                ^~~~~~~~~~~~~
kis_ppilogfile.cc:97:63: error: invalid use of ‘this’ in non-member function
  packetchain->RegisterHandler(&KisPPILogfile::packet_handler, this, CHAINPOS_LOGGING, -100);
                                                               ^~~~
kis_ppilogfile.cc: At global scope:
kis_ppilogfile.cc:102:6: error: ‘KisPPILogfile’ has not been declared
 void KisPPILogfile::Log_Close() {
      ^~~~~~~~~~~~~
kis_ppilogfile.cc: In function ‘void Log_Close()’:
kis_ppilogfile.cc:103:24: error: ‘log_mutex’ was not declared in this scope
     local_locker lock(&log_mutex);
                        ^~~~~~~~~
kis_ppilogfile.cc:105:5: error: ‘set_int_log_open’ was not declared in this scope
     set_int_log_open(false);
     ^~~~~~~~~~~~~~~~
kis_ppilogfile.cc:105:5: note: suggested alternative: ‘sqlite3_blob_open’
     set_int_log_open(false);
     ^~~~~~~~~~~~~~~~
     sqlite3_blob_open
kis_ppilogfile.cc:110:37: error: ‘KisPPILogfile’ has not been declared
         packetchain->RemoveHandler(&KisPPILogfile::packet_handler, CHAINPOS_LOGGING);
                                     ^~~~~~~~~~~~~
kis_ppilogfile.cc:113:9: error: ‘dumper’ was not declared in this scope
     if (dumper != NULL) {
         ^~~~~~
kis_ppilogfile.cc:113:9: note: suggested alternative: ‘dup3’
     if (dumper != NULL) {
         ^~~~~~
         dup3
kis_ppilogfile.cc:114:9: error: ‘pcap_dump_flush’ was not declared in this scope
         pcap_dump_flush(dumper);
         ^~~~~~~~~~~~~~~
kis_ppilogfile.cc:115:9: error: ‘pcap_dump_close’ was not declared in this scope
         pcap_dump_close(dumper);
         ^~~~~~~~~~~~~~~
kis_ppilogfile.cc:118:9: error: ‘dumpfile’ was not declared in this scope
     if (dumpfile != NULL) {
         ^~~~~~~~
kis_ppilogfile.cc:118:9: note: suggested alternative: ‘tmpfile’
     if (dumpfile != NULL) {
         ^~~~~~~~
         tmpfile
kis_ppilogfile.cc:119:9: error: ‘pcap_close’ was not declared in this scope
         pcap_close(dumpfile);
         ^~~~~~~~~~
kis_ppilogfile.cc:119:9: note: suggested alternative: ‘pclose’
         pcap_close(dumpfile);
         ^~~~~~~~~~
         pclose
kis_ppilogfile.cc:122:2: error: ‘dumper’ was not declared in this scope
  dumper = NULL;
  ^~~~~~
kis_ppilogfile.cc:122:2: note: suggested alternative: ‘dup3’
  dumper = NULL;
  ^~~~~~
  dup3
kis_ppilogfile.cc:123:2: error: ‘dumpfile’ was not declared in this scope
  dumpfile = NULL;
  ^~~~~~~~
kis_ppilogfile.cc:123:2: note: suggested alternative: ‘tmpfile’
  dumpfile = NULL;
  ^~~~~~~~
  tmpfile
kis_ppilogfile.cc:124:5: error: ‘dump_filep’ was not declared in this scope
     dump_filep = NULL;
     ^~~~~~~~~~
kis_ppilogfile.cc:124:5: note: suggested alternative: ‘tmpfile’
     dump_filep = NULL;
     ^~~~~~~~~~
     tmpfile
kis_ppilogfile.cc: At global scope:
kis_ppilogfile.cc:128:1: error: ‘KisPPILogfile’ does not name a type
 KisPPILogfile::~KisPPILogfile() {
 ^~~~~~~~~~~~~
kis_ppilogfile.cc:132:6: error: ‘KisPPILogfile’ has not been declared
 void KisPPILogfile::RegisterPPICallback(dumpfile_ppi_cb in_cb, void *in_aux) {
      ^~~~~~~~~~~~~
kis_ppilogfile.cc:132:41: error: variable or field ‘RegisterPPICallback’ declared void
 void KisPPILogfile::RegisterPPICallback(dumpfile_ppi_cb in_cb, void *in_aux) {
                                         ^~~~~~~~~~~~~~~
kis_ppilogfile.cc:132:41: error: ‘dumpfile_ppi_cb’ was not declared in this scope
kis_ppilogfile.cc:132:64: error: expected primary-expression before ‘void’
 void KisPPILogfile::RegisterPPICallback(dumpfile_ppi_cb in_cb, void *in_aux) {
                                                                ^~~~
kis_ppilogfile.cc:145:6: error: ‘KisPPILogfile’ has not been declared
 void KisPPILogfile::RemovePPICallback(dumpfile_ppi_cb in_cb, void *in_aux) {
      ^~~~~~~~~~~~~
kis_ppilogfile.cc:145:39: error: variable or field ‘RemovePPICallback’ declared void
 void KisPPILogfile::RemovePPICallback(dumpfile_ppi_cb in_cb, void *in_aux) {
                                       ^~~~~~~~~~~~~~~
kis_ppilogfile.cc:145:39: error: ‘dumpfile_ppi_cb’ was not declared in this scope
kis_ppilogfile.cc:145:62: error: expected primary-expression before ‘void’
 void KisPPILogfile::RemovePPICallback(dumpfile_ppi_cb in_cb, void *in_aux) {
                                                              ^~~~
kis_ppilogfile.cc:154:5: error: ‘KisPPILogfile’ has not been declared
 int KisPPILogfile::packet_handler(CHAINCALL_PARMS) {
     ^~~~~~~~~~~~~
kis_ppilogfile.cc: In function ‘int packet_handler(GlobalRegistry*, void*, kis_packet*)’:
kis_ppilogfile.cc:155:5: error: ‘KisPPILogfile’ was not declared in this scope
     KisPPILogfile *ppilog = (KisPPILogfile *) auxdata;
     ^~~~~~~~~~~~~
kis_ppilogfile.cc:155:20: error: ‘ppilog’ was not declared in this scope
     KisPPILogfile *ppilog = (KisPPILogfile *) auxdata;
                    ^~~~~~
kis_ppilogfile.cc:155:20: note: suggested alternative: ‘__ilogb’
     KisPPILogfile *ppilog = (KisPPILogfile *) auxdata;
                    ^~~~~~
                    __ilogb
kis_ppilogfile.cc:155:45: error: expected primary-expression before ‘)’ token
     KisPPILogfile *ppilog = (KisPPILogfile *) auxdata;
                                             ^
kis_ppilogfile.cc:451:24: error: aggregate ‘packet_handler(GlobalRegistry*, void*, kis_packet*)::pcap_pkthdr wh’ has incomplete type and cannot be defined
     struct pcap_pkthdr wh;
                        ^~
kis_ppilogfile.cc:459:9: error: ‘pcap_dump’ was not declared in this scope
         pcap_dump((u_char *) ppilog->dumper, &wh, dump_data);
         ^~~~~~~~~
Makefile:588: recipe for target 'kis_ppilogfile.cc.o' failed
make: *** [kis_ppilogfile.cc.o] Error 1

kismetwireless, 2019-07-01:
Platform?  Distribution?  Compiler version?  Output from configure? I don't have a system on which this doesn't build properly, so I'm not sure what you're missing.

aqua95, 2019-07-01:
gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1) |  Output from configure? how do i find this?

aqua95, 2019-07-01:
Configuration complete: 
         Compiling for: linux-gnu (x86_64)
           C++ Library: stdc++
   Installing as group: root
       Installing into: /usr/local
          Setuid group: kismet
   Linux WEXT capture : yes
        Prelude  SIEM : no
   PCRE Regex Filters : yes
          pcap capture: yes
       airpcap control: n/a (only Cygwin/Win32)
        PPI log format: no - PPI logging unavailable, upgrade libpcap
LibCapability (enhanced
   privilege dropping): yes
   OSX/Darwin capture : n/a (only OSX/Darwin)
         Linux Netlink: no (will not be able to make mac80211 vaps)
  Linux NetworkManager: yes (libnm NetworkManager control)
       Linux Bluetooth: yes
         nRF MOuseJack: yes
        Python Modules: yes
       RTL-SDR RTL_433: yes
       RTL-SDR RTLAMR: yes
      RTL-SDR RTLADSB: yes
      Freaklabs Zigbee: yes
 lm-sensors monitoring: no (will not be able to monitor system temperature, etc)
        Built-in Debug: partial - Missing libdw or libbfd will not print full stacks on crash

kismetwireless, 2019-07-01:
Yep; looks like your libpcap lacks ppi support.   I don't know how that's
happened since libpcap in ubuntu definitely has it, and has done for a
decade.  Make sure you have all the current stuff installed and the
required list of packages installed from the docs:

https://www.kismetwireless.net/docs/readme/quickstart/

Kismet will build fine on ubuntu 18.04 so you've unfortunately got
something wacky going on.

aqua95, 2019-07-01:
Ok, yeah that was the issue. I installed the latest version of lipcap and ran the configure file again and it worked. Thanks!

kismetwireless, 2019-07-01:
Awesome; I've also updated the autoconf to make it non-optional so it ought
to blow up w/ a sane error now, too.

closed 2019-07-01 by kismetwireless