kismetwireless/kismet #155: error in sudo KIS_SRC_DIR=/usr/src/kismet make - Makefile:43: recipe for target 'packetsource_ubertooth.o' failed

aqua95, 2019-07-01:
g++ -std=gnu++17 -fopenmp  -Wall -g -I. -g -O2 -O3 -pthread -DKS_STR_ENCODING_NONE -I/usr/src/kismet -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c packetsource_ubertooth.cc -o packetsource_ubertooth.o 
In file included from /usr/src/kismet/gpsdclient.h:23:0,
                 from /usr/src/kismet/packetsource.h:30,
                 from packetsource_ubertooth.cc:32:
/usr/src/kismet/clinetframework.h: In constructor ‘ClientFramework::ClientFramework(GlobalRegistry*)’:
/usr/src/kismet/clinetframework.h:150:14: error: ‘class GlobalRegistry’ has no member named ‘RegisterPollableSubsys’; did you mean ‘RegisterUsageFunc’?
   globalreg->RegisterPollableSubsys(this);
              ^~~~~~~~~~~~~~~~~~~~~~
              RegisterUsageFunc
/usr/src/kismet/clinetframework.h: In destructor ‘virtual ClientFramework::~ClientFramework()’:
/usr/src/kismet/clinetframework.h:154:14: error: ‘class GlobalRegistry’ has no member named ‘RemovePollableSubsys’
   globalreg->RemovePollableSubsys(this);
              ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/src/kismet/kis_netframe.h:26:0,
                 from /usr/src/kismet/gpsdclient.h:25,
                 from /usr/src/kismet/packetsource.h:30,
                 from packetsource_ubertooth.cc:32:
/usr/src/kismet/netframework.h: At global scope:
/usr/src/kismet/netframework.h:122:35: error: ‘vector’ has not been declared
     virtual int FetchClientVector(vector<int> *ret_vec);
                                   ^~~~~~
/usr/src/kismet/netframework.h:122:41: error: expected ‘,’ or ‘...’ before ‘<’ token
     virtual int FetchClientVector(vector<int> *ret_vec);
                                         ^
/usr/src/kismet/netframework.h:129:10: error: ‘string’ does not name a type; did you mean ‘stdin’?
  virtual string GetRemoteAddr(int in_fd) = 0;
          ^~~~~~
          stdin
/usr/src/kismet/netframework.h:161:5: error: ‘map’ does not name a type
     map<int, RingBuffer *> read_buf_map;
     ^~~
/usr/src/kismet/netframework.h:163:5: error: ‘map’ does not name a type
     map<int, RingBuffer *> write_buf_map;
     ^~~
/usr/src/kismet/netframework.h: In member function ‘virtual int NetworkServer::FetchNumClients()’:
/usr/src/kismet/netframework.h:96:22: error: ‘write_buf_map’ was not declared in this scope
         return (int) write_buf_map.size();
                      ^~~~~~~~~~~~~
In file included from /usr/src/kismet/gpsdclient.h:25:0,
                 from /usr/src/kismet/packetsource.h:30,
                 from packetsource_ubertooth.cc:32:
/usr/src/kismet/kis_netframe.h: At global scope:
/usr/src/kismet/kis_netframe.h:45:23: error: ‘string’ has not been declared
  void Cache(int in_f, string in_val) {
                       ^~~~~~
/usr/src/kismet/kis_netframe.h:50:2: error: ‘string’ does not name a type; did you mean ‘stdin’?
  string GetCache(int in_f) {
  ^~~~~~
  stdin
/usr/src/kismet/kis_netframe.h:73:2: error: ‘vector’ does not name a type; did you mean ‘herror’?
  vector<string> field_cache;
  ^~~~~~
  herror
/usr/src/kismet/kis_netframe.h:74:2: error: ‘vector’ does not name a type; did you mean ‘herror’?
  vector<int> field_filled;
  ^~~~~~
  herror
/usr/src/kismet/kis_netframe.h: In member function ‘int kis_protocol_cache::Filled(int)’:
/usr/src/kismet/kis_netframe.h:43:10: error: ‘field_filled’ was not declared in this scope
   return field_filled[in_f];
          ^~~~~~~~~~~~
/usr/src/kismet/kis_netframe.h: In member function ‘void kis_protocol_cache::Cache(int, int)’:
/usr/src/kismet/kis_netframe.h:47:3: error: ‘field_cache’ was not declared in this scope
   field_cache[in_f] = in_val;
   ^~~~~~~~~~~
/usr/src/kismet/kis_netframe.h:48:3: error: ‘field_filled’ was not declared in this scope
   field_filled[in_f] = 1;
   ^~~~~~~~~~~~
/usr/src/kismet/kis_netframe.h: In member function ‘int kis_protocol_cache::CacheResize(int)’:
/usr/src/kismet/kis_netframe.h:60:3: error: ‘field_cache’ was not declared in this scope
   field_cache.resize(in_f + 1, string(""));
..... and more lines

kismetwireless, 2019-07-01:
> virtual int FetchClientVector(vector *ret_vec);
> vector field_cache;
> vector field_filled;

Looks like you're trying to compile the old ubertooth plugin or something;
that's all old code which is unsupported at this point.  There is currently
no direct ubertooth support in the new kismet versions, but it will likely
return at some point in the future.

aqua95, 2019-07-01:
Alright how could we make kismet and ubertooth work together?

aqua95, 2019-07-01:
or is there anything else which could help me integrate ubertooth and kismet?

kismetwireless, 2019-07-02:
Currently there isn't integration of modern kismet and ubertooth; it's on the list but no ETA.

closed 2019-07-02 by kismetwireless