kismetwireless/kismet #157: Not able to start interface over command line (-c)

JK2210, 2019-07-03:
When I try to start kismet with a specific interface (`kismet -c source:wlan1`) I get an error:

```
ERROR: Unable to find driver for 'source:wlan1'.  Make sure that any 
       required plugins are loaded, the interface is available, and any 
       required Kismet helper packages are installed.
ERROR: Data source 'source:wlan1' failed to launch: Unable to find driver 
       for 'source:wlan1'.  Make sure that any required plugins are 
       loaded, the interface is available, and any required Kismet helper 
       packages are installed.
```
But when I start the interface over the web UI it works great.

I am using Kismet 2019-05-02 on a current raspbian.

Do you have any idea, why this only works in web UI and not over command line?

kismetwireless, 2019-07-03:
For the command line it's just '-c wlan1' or for more advanced ones, '-c
wlan1:name=foo' and so on

JK2210, 2019-07-03:
Sorry, my mistake!

closed 2019-07-03 by JK2210

dandelionnone, 2020-06-12:
I get the same error when starting with 'kismet -c wlan0'. The interface is up, and in monitor mode. I verified in Wireshark.

dandelionnone, 2020-06-12:
Never mind, I realized I had not added myself to the kismet group". Works perfectly now.

SMFHS, 2023-03-25:
> Never mind, I realized I had not added myself to the kismet group". Works perfectly now.

How do you do that?

Raviolibutthole, 2024-05-03:
> > Never mind, I realized I had not added myself to the kismet group". Works perfectly now.
> 
> How do you do that?

>>>sudo usermod -aG kismet 'your user'