kismetwireless/kismet #164: capture_sdr_rtladsb/setup.py is pulling down numpy 1.17.0rc2

drewlander, 2019-07-18:
when checking out the latest git, I attempted to build on OSX 10.14.5 and also Fedora 30 and ran into the same problem.
```
File "/tmp/easy_install-CW458J/numpy-1.17.0rc2/setup.py", line 31, in <module>
    
RuntimeError: Python version >= 3.5 required.
```
It seems that the latest version of numpy has dropped python 2.x support [source](https://github.com/numpy/numpy/blob/master/setup.py#L31)
If I pip install  pip install numpy==1.16.4 and then build kismet it works flawlessly

looks like it is being pulled in from capture_sdr_rtladsb/setup.py (version is not pinned)

Is this something that needs addressed or did I miss a setup step somewhere (or is this already documented and I missed it)?

Thanks!

kismetwireless, 2019-07-18:
I'll probably transition all the capture stuff to python 3 for the next
release (in all the distros that support it). For now you can probably use
the system package for python2 numpy which on deb systems should just be
python-numpy

drewlander, 2019-07-18:
Excellent . Just wanted to make sure it was something that was known or if I missed something obvious in the documentation. Thank you.

kismetwireless, 2019-07-18:
The latest git should now use python3 by default; i'd already done most of
the work porting the kismet-external framework to py3; the actual launcher
tools are now setuptools entrypoints to autogen py2 or py3 variants; if you
want to give it a spin and see how it goes.

drewlander, 2019-07-19:
I removed numpy from python installation, rebuilt, and it failed (As expected)
I did a fresh clone from master and it built without a problem on both OSX and Fedora 30.
It also is working as expected as well. The issue seems to be fixed, thank you for the quick fix!

closed 2019-07-20 by kismetwireless