kismetwireless/kismet #166: Kismet Server Crashes w/ Remote Capture

sketchybinary, 2019-07-31:
I'm using commit 409cc5d02c06b2a6f945a6110cc3ff84a6b9208a and I have 8 remote linux_cap_wifi sources feeding the Kismet Server.

`Stack trace (most recent call last) in thread 80:
Stack trace (most recent call last):
#12   Object ", at 0xffffffffffffffff, in 
#15   Object ", at 0xffffffffffffffff, in 
#14   Object "/lib/x86_64-linux-gnu/libc-2.27.so, at 0x7f00e29d788e, in __clone
#13   Object "/lib/x86_64-linux-gnu/libpthread-2.27.so, at 0x7f00e3c756da, in start_thread
#12   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25, at 0x7f00e354966e, in 
#11   Object "/usr/local/bin/kismet, at 0x559731a149a9, in _start
#10   Object "/lib/x86_64-linux-gnu/libc-2.27.so, at 0x7f00e28d7b96, in __libc_start_main
#11 | Source "/usr/include/c++/7/thread", line 186, in operator()
    |   185:    void
    | > 186:    _M_run() { _M_func(); }
    |   187:       };
    | Source "/usr/include/c++/7/thread", line 243, in _M_invoke<0>
    |   241:    noexcept(noexcept(std::declval<_Invoker&>()._M_invoke(_Indices())))
    |   242:    -> decltype(std::declval<_Invoker&>()._M_invoke(_Indices()))
    | > 243:    { return _M_invoke(_Indices()); }
    |   244:       };
    | Source "/usr/include/c++/7/thread", line 234, in __invoke<Datasourcetracker::open_remote_datasource(dst_incoming_remote*, const string&, const string&, const uuid&, std::shared_ptr<BufferHandlerGeneric>)::<lambda()> >
    |   232:      noexcept(noexcept(std::__invoke(_S_declval<_Ind>()...)))
    |   233:      -> decltype(std::__invoke(_S_declval<_Ind>()...))
    | > 234:      { return std::__invoke(std::get<_Ind>(std::move(_M_t))...); }
    |   235: 
    |   236:    using _Indices
    | Source "/usr/include/c++/7/bits/invoke.h", line 95, in __invoke_impl<void, Datasourcetracker::open_remote_datasource(dst_incoming_remote*, const string&, const string&, const uuid&, std::shared_ptr<BufferHandlerGeneric>)::<lambda()> >
    |    93:       using __type = typename __result::type;
    |    94:       using __tag = typename __result::__invoke_type;
    | >  95:       return std::__invoke_impl<__type>(__tag{}, std::forward<_Callable>(__fn),
    |    96:                                    std::forward<_Args>(__args)...);
    |    97:     }
    | Source "/usr/include/c++/7/bits/invoke.h", line 60, in operator()
    |    58:     constexpr _Res
    |    59:     __invoke_impl(__invoke_other, _Fn&& __f, _Args&&... __args)
    | >  60:     { return std::forward<_Fn>(__f)(std::forward<_Args>(__args)...); }
    |    61: 
    |    62:   template<typename _Res, typename _MemFun, typename _Tp, typename... _Args>
      Source "/opt/kismet/datasourcetracker.cc", line 1115, in _M_run
#10   Source "/opt/kismet/kis_datasource.cc", line 354, in connect_remote
#9    Source "/opt/kismet/kismet_server.cc", line 954, in main
#8  | Source "/opt/kismet/pollabletracker.cc", line 108, in ProcessPollableSelect
      Source "/opt/kismet/pollabletracker.cc", line 130, in Selectloop
#9    Source "/opt/kismet/kis_external.cc", line 64, in connect_buffer
#7    Source "/opt/kismet/tcpserver2.cc", line 226, in Poll
#8    Source "/opt/kismet/buffer_handler.cc", line 360, in SetReadBufferInterface
#6    Source "/opt/kismet/buffer_handler.cc", line 306, in CommitReadBufferData
#7    Source "/opt/kismet/kis_external.cc", line 113, in BufferAvailable
#6    Source "/opt/kismet/ringbuf2.cc", line 178, in peek
#5    Source "/opt/kismet/kis_mutex.h", line 631, in local_eol_locker
#5    Source "/opt/kismet/kis_mutex.h", line 396, in local_locker
#4    Source "/opt/kismet/kis_mutex.h", line 150, in try_lock_for
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25, at 0x7f00e351ed23, in __cxa_throw
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25, at 0x7f00e351eaf0, in std::terminate()
#1    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25, at 0x7f00e351eab5, in 
#4    Source "/opt/kismet/kis_mutex.h", line 150, in try_lock_for
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25, at 0x7f00e351ed23, in __cxa_throw
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25, at 0x7f00e351eaf0, in std::terminate()
#1    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25, at 0x7f00e351eab5, in 
#0    Source "/opt/kismet/kismet_server.cc", line 352, in TerminationHandler
`

sketchybinary, 2019-07-31:
I was able to recreate this issue with a build that came off of master as well

kismetwireless, 2019-07-31:
Anything specific you're doing to trigger that?

kismetwireless, 2019-07-31:
ok; i got it to trigger, too.  For now, i'm reverting the new polling pool code I added this week; that seems to be the root cause of it.  I'll revisit it later if need be.

sketchybinary, 2019-07-31:
Wow, that was fast. Thanks for all of your great work.