kismetwireless/kismet #167: kis_log_packets=false does not work (or kis_log_messages=false, or kis_log_packet_filter_default=block)

designer2k2, 2019-08-01:
Hello,
im trying to reduce the memory/space consumption on a device tracking only system.
So the packet data is not needed.

But so far this configs have no effect, i still have all the packet/message in the database, and no reduced memory usage.

i have this in my kismet_site.conf:

kis_log_packets=false 
kis_log_messages=false
kis_log_packet_filter_default=block

(there is also gpsd and source settings, but they work)

im running kismet 2019.07.R2 on a rpi 3b with kali 64bit

Is there some setting im missing or is this a bug?

kismetwireless, 2019-08-01:
It works as expected here; setting kis_log_packets=false shows no packets
in a sample log.  the code path for this one is really simple so I'm not
sure where it's going wrong for you; there's only one place it can attach
to the packet chain for logging to the kismetdb file, and it's in that
test...

You won't, fwiw, see any reduced memory, only reduced database logs.

designer2k2, 2019-08-02:
I assumed that the *.kismet ist the "log" this is about?

Here is a short sample of what i get with the settings from above:
[Kismet-20190801-21-08-46-1.zip](https://github.com/kismetwireless/kismet/files/3459475/Kismet-20190801-21-08-46-1.zip)

It contains all messages/packets, i was expecting something like when i use the kismetdb_strip_packet_content tool. Or actually i was hoping to have empty packets/messages tables.

kismetwireless, 2019-08-02:
Yep, that log has packets in it - but I can't seem to replicate that
behavior; setting "kis_log_packets=false" in kismet_site.conf works as
expected on all my tests, and the packets table is empty.

kismetwireless, 2019-08-02:
I just put an explicit "packets won't be saved" print in the git-master
code if you want to either try compiling that or waiting until tomorrows
nightly build; there's no logic change, just a confirmatory output.

designer2k2, 2019-08-02:
ok, i found the error, it was on my side.  

I created the kismet_site.conf with winscp, resulting in CRLF linefeeds.  That made the conf file nearly useless.  Somehow the source and gps went ok, but everything else got ignored.

I now changed the CRLF to LF and it works as expected, no packets and no messages!

Maybe there is a way that kismet warns if a "bad" conf encoding is used?  But i think thats a low priority thing...

kismetwireless, 2019-08-02:
Oh.  That's interesting.  Yep, I can see how that would happen.

It may not be super high priority, but it also might not be very complex to
either detect that, or support \r\n and \n both.  I don't think i'd have
thought of looking for that; good find!

designer2k2, 2019-08-07:
may i ask why i dont see a reduced memory consumption? 
I can confirm that with this options the resulting database is much smaller, but the memory footprint is about unchanged.

And what else could i do to reduce the memory consumption significantly?

kismetwireless, 2019-08-07:
Logging (or not logging) packets won't do anything for RAM because packets
are purely ephemeral content; they're processed and logged to disk.

All the memory tuning options are documented at
https://www.kismetwireless.net/docs/readme/performance_and_memory/

>

designer2k2, 2019-08-07:
thank you!

i will then close this issue here.

closed 2019-08-07 by designer2k2